neu-rah/ArduinoMenu #174: timeout is not working

lucky62, 2018-04-29:
I have `nav.timeOut = 10;` in the code but menu never goes to idle state.
What conditions must be satisfied to go into idle state after time out?

neu-rah, 2018-04-29:
only when **output** is quiet for more than `nav.timeOut` seconds, so if you have visible fields with values changing it wont timeout, otherwise its a bug

lucky62, 2018-04-30:
Most probably it is a bug...
timeOut is checked only in navRoot::checked() method. But in which cases is this method called?

neu-rah, 2018-04-30:
its called every poll, and it checks if the output need to be redraw

neu-rah, 2018-04-30:
what is your hardware?
I'm using timeOut on a current sketch using u8g2 + ssd1306 with serial and web interface and it behaves ok.

can you past the outputs init code?

either here on at https://gitter.im/ArduinoMenu/Lobby

lucky62, 2018-04-30:
Hardware: Arduino Pro Mini clone, LCD 16x2 with I2C adapter, rotary encoder with push button

Lcd, Encoder & Menu initialization:
```
// LCD Display
LiquidCrystal_I2C lcd(LCD_I2C_ADDR, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address and pinout

ClickEncoder clickEncoder(ENC_A,ENC_B,ENC_BTN,4);
void timerIsr() {clickEncoder.service();}

ClickEncoderStream encStream(clickEncoder,1);
MENU_INPUTS(menu_in,&encStream);

#define MAX_DEPTH 2
MENU_OUTPUTS(menu_out,MAX_DEPTH
  ,LCD_OUT(lcd,{0,0,16,2})
  ,NONE
);

// Definícia Menu
TOGGLE(s.lcdBacklight,menuLcdBacklight,"Backlight: ",doNothing,noEvent,noStyle//,doExit,enterEvent,noStyle
  ,VALUE("On",HIGH,doNothing,noEvent)
  ,VALUE("Off",LOW,doNothing,noEvent)
);

TOGGLE(s.active[0],menusec1Active,"S1 Active: ",doNothing,noEvent,noStyle//,doExit,enterEvent,noStyle
  ,VALUE("Yes",HIGH,doNothing,noEvent)
  ,VALUE("No",LOW,doNothing,noEvent)
);

TOGGLE(s.active[1],menusec2Active,"S2 Active: ",doNothing,noEvent,noStyle//,doExit,enterEvent,noStyle
  ,VALUE("Yes",HIGH,doNothing,noEvent)
  ,VALUE("No",LOW,doNothing,noEvent)
);

TOGGLE(s.active[2],menusec3Active,"S3 Active: ",doNothing,noEvent,noStyle//,doExit,enterEvent,noStyle
  ,VALUE("Yes",HIGH,doNothing,noEvent)
  ,VALUE("No",LOW,doNothing,noEvent)
);

MENU(menuSec1,"Sekcia 1",doNothing,anyEvent,noStyle
  ,SUBMENU(menusec1Active)
  ,FIELD(s.minHum[0],"S1 Min.Hum.","",0,1000,100,10,doNothing,noEvent,wrapStyle)
  ,EXIT("Exit")
);

MENU(menuSec2,"Sekcia 2",doNothing,anyEvent,noStyle
  ,SUBMENU(menusec2Active)
  ,FIELD(s.minHum[1],"S2 Min.Hum.","",0,1000,100,10,doNothing,noEvent,wrapStyle)
  ,EXIT("Exit")
);

MENU(menuSec3,"Sekcia 3",doNothing,anyEvent,noStyle
  ,SUBMENU(menusec3Active)
  ,FIELD(s.minHum[2],"S3 Min.Hum.","",0,1000,100,10,doNothing,noEvent,wrapStyle)
  ,EXIT("Exit")
);

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(menuLcdBacklight)
  ,SUBMENU(menuSec1)
  ,SUBMENU(menuSec2)
  ,SUBMENU(menuSec3)
  ,EXIT("EXIT")
);

NAVROOT(nav,mainMenu,MAX_DEPTH,menu_in,menu_out);//the navigation root object

result idle(menuOut& o,idleEvent e) {
  switch(e) {
    case idleStart: menuActive = false; Serial.println("Menu Idle Start..."); saveSettings(); break;
    case idling: break;
    case idleEnd: menuActive = true; Serial.println("Menu Idle Stop."); break;
  }
  return proceed;
}
```

Later in setup():
```
  nav.idleTask=idle;//point a function to be used when menu is suspended
  nav.timeOut = 10;
  nav.showTitle=false;
  nav.exit();

```
Menu is activated by pressing rotary encoder, navigation works ok, I can exit the menu by EXIT but not by timeOut.

neu-rah, 2018-05-20:
confirmed, this was a recent feature and was working only on u8g2 due to the direct check of chenged state, fixed.
tested it on i2c lcd and should work for all devices
thanks for drawing my attention to it :+1:

closed 2018-05-20 by neu-rah