neu-rah/ArduinoMenu #175: `changed` report bug

neu-rah, 2018-05-20:
confirmed BUG when using I2c lcd, i think its not specific to I2c LCD but noticed that toggle field on sub menu does not update when changed (have to scroll to see change). Using a 16x2 LCD with no menu title, possibly related to changed function logic bug (on no title device)... checking

reported by @d21d3q on gitter
thanks!

neu-rah, 2018-05-20:
this is not related to title presence, on sub menu with title we have navigation scroll issue...
this is a severe bug

neu-rah, 2018-05-20:
fixed, using new `changed`/`clearChanged` unified code
this will require a close verification as this functions are crucial to all menu navigation...
tester are very welcome, thank you.

closed 2018-05-20 by neu-rah

d21d3q, 2018-06-13:
I've reported on gitter that LCD is not refreshing on value update while serial out is being refreshed.
I've tested version 4.8.2 (pulled from arduino ide) with following (minimal) code:
```C
#include <Arduino.h>
#include <menu.h>
#include <menuIO/liquidCrystalOut.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>
#include <menuIO/chainStream.h>

using namespace Menu;

// LCD /////////////////////////////////////////
#define RS A0
//#define RW 3
#define EN A1
LiquidCrystal lcd(RS, /*RW,*/ EN, A2, A3, A4, A5);

//input from the encoder + encoder button + serial
serialIn serial(Serial);
menuIn* inputsList[]={&serial};
chainStream<1> in(inputsList);//3 is the number of inputs

int test=0;

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,FIELD(test, "test", " units",0,0,0,0,doNothing,noEvent,wrapStyle)
  ,EXIT("<Back")
);

#define MAX_DEPTH 1

MENU_OUTPUTS(out, MAX_DEPTH
  ,LIQUIDCRYSTAL_OUT(lcd,{0,0,16,2})
  ,SERIAL_OUT(Serial)
  ,NONE
);
NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);//the navigation root object

void setup() {
//  pinMode(encBtn,INPUT_PULLUP);
  Serial.begin(115200);
  while(!Serial);
  Serial.println("Arduino Menu Library");Serial.flush();
//  encoder.begin();
  lcd.begin(16,2);
}

unsigned long prev;

void loop() {
  nav.poll();
  if (prev + 1000 < millis()) {
    prev = millis();
    test++;
  }
  delay(100);//simulate a delay as if other tasks are running
}
```
and LCD is being refreshed as expected, but now serial out is not being refreshed.

neu-rah, 2018-06-15:
```c++
MENU_OUTPUTS(out, MAX_DEPTH
  ,LIQUIDCRYSTAL_OUT(lcd,{0,0,16,2})
  ,SERIAL_OUT(Serial)
  ,NONE
);
```
output order has any influence?
and NONE is not needed (only needed for macro has it expect a minimal of two elements

d21d3q, 2018-06-17:
yes, first one is being refreshed, second one not.