neu-rah/ArduinoMenu #176: inverted colors on different boards?

szmolnar, 2018-05-20:
Hi there! 

I'm playing with u8g2 example code.
By compiling the unmodified example code with constructor "U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ 4, /* data=*/ 5);
" on a WEMOS Lolin32 is just working fine.
By compiling the same code for an arduino nano I'm getting inverted colors. (background is white)

Am I doing something wrong? Any idea?

Thanks!

neu-rah, 2018-05-21:
interesting... gota check that
you can invert the color table, but that was not expected

neu-rah, 2018-05-21:
that reminds me the led case, esp boards have a led connected to the positive making the led state inverse, wonder if something similar can occur with the oled, is it the builtin oled version?

szmolnar, 2018-05-21:
Yes, I’m using the one with built in display. 
But please pay attention that everything is ok on that board. 
(Original code outputs black background, white characters, so it’s ok)

I’m getting inverted colors with exactly same code, also same constructor and by using an Arduino Nano board with wired ssd1306.

I was trying to invert it from code but it is not really clear for me at the moment how color table is working.
“
//each color is in the format:
// {{disabled normal,disabled selected},{enabled normal,enabled selected, enabled editing}}
// this is a monochromatic color table
const colorDef<uint8_t> colors[]={
{{0,0},{0,1,1}},//bgColor
{{1,1},{1,0,0}},//fgColor
{{1,1},{1,0,0}},//valColor
{{1,1},{1,0,0}},//unitColor
{{0,1},{0,0,1}},//cursorColor
{{1,1},{1,0,0}},//titleColor
};

“

I suppose 0=black, 1=white.
In my understanding it should invert at least the ,"Main menu" text if i  invert the lines below but it does not work. 
“
{{0,0},{0,1,1}},//bgColor
{{1,1},{1,0,0}},//fgColor
“
to
“
{{1,1},{1,0,0}},//bgColor
{{0,0},{0,1,1}},//fgColor
“

Also I wish to mention that same configurations are working fine with U8g2 example codes from olikraus. ( same constructor, black background and white characters on both boards)

Thanks!

neu-rah, 2018-05-21:
hi!

thanks for your quick response.

>Also I wish to mention that same configurations are working fine with U8g2 example codes from olikraus. ( same constructor, black background and white characters on both boards)

this part is important and rules out some electronic inversion on the board...

I have one esp32 with integrated oled, will try on it

thank you :D

szmolnar, 2018-05-21:
Hi!

hmm, why do you think is an electronic issue?
U8g2 example code from olikraus is working fine on esp32 board with integrated OLED and also on my nano with same wiring, same code.
At the other hand your u2g8 example is only working perfectly on the esp32 board with integrated OLED but not on my nano with same wiring, same code.

So in my opinion there shouldn’t be any board wiring issue otherwise olikraus’s example should also behave as yours.

Anyway I hope you will figure out whats wrong.

Do you have an example code for uno or nano (Atmega328P) boards with ssd1306 I could give  try?
Or any hint why sw color inverting is not working for me?

Thanks!

neu-rah, 2018-05-21:
**rules out** mean its not

neu-rah, 2018-05-21:
i'm pushing a new example for lolin32 board with onboard oled

and this video with the result https://youtu.be/i49h7FmrsBk

using a board like this one: https://www.ebay.com/itm/ESP32-OLED-WiFi-Bluetooth-WROOM-32-ESP-32-ESP-32S-ESP8266-Wemos-Lolin32-arduin/123144901146?hash=item1cac02121a:g:m7IAAOSwmNxadGnB

not sure if we have the same board or if we have the same result...

neu-rah, 2018-05-21:
P.S. will do an assembly using nano and oled to check it also ;)

neu-rah, 2018-05-21:
confirming that nano and u8g2+ssd1306 example create an absolute mess, i'm still sorting it out...

neu-rah, 2018-05-21:
ok, fixed it
the reason for the difference on color tables was because avr's (like nano) use progmem to store the data and save ram, unfortunately the color table was missing a MEMMODE attribute, that translates to PROGMEM on AVR's.

as ESP is not affected by it, it was getting the right color table while the AVR was picking some garbage while looking into flash memory for the color table.

thanks for noticing it and opening this issue.

please reopen if some problem persists, or open a new one if you find any other problem.

corrections are pushed to repo.

closed 2018-05-21 by neu-rah

szmolnar, 2018-05-21:
Yep, it’s OK now!
Thanks for your support!

neu-rah, 2018-05-21:
you're welcome! and thanks for your patience and collaboration, testing this huge combination of hardware is not an easy task so i'm grateful to all people that use and test this :D

you are also welcome to join us on gitter https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link