neu-rah/ArduinoMenu #179: idleChanged=true does not call my idle function constantly

me21, 2018-05-29:
As stated in https://github.com/neu-rah/ArduinoMenu/wiki/Idling#the-hard-way, setting `nav.idleChanged = true;` should lead to my function being called constantly after `idleOn(myFunction);` call.
But it doesn't.
My function is called once with `idleStart` event, and then nothing.
I think it's due to https://github.com/neu-rah/ArduinoMenu/blob/9d0a6525b7c7f3bb2838fe5ecd409d1eed8afd2a/src/io.h#L215
I use LCD screen, and its style is `minimalRedraw` by default. That's why my idle function is not called constantly.

neu-rah, 2018-05-30:
are you setting it inside your idle function, right?

me21, 2018-05-30:
Can you explain what different styles mean?

neu-rah, 2018-05-30:
```c++
        enum styles {
          none=0<<0, // default serialOut
          redraw=1<<0,// changing one part implies printing all visible
          minimalRedraw=1<<1, // can draw part of the screen
          drawNumIndex=1<<2, // numerical index should be displayed
          usePreview=1<<3, // reserve one panel for preview
          expandEnums=1<<4, // enumerated typed should be printed along with the option (web interface)
          rasterDraw=1<<5 // will request multiple draws to complete a screen
        } style;
```

you are also welcome to join us on gitter https://gitter.im/ArduinoMenu/Lobby

closed 2018-07-18 by neu-rah