neu-rah/ArduinoMenu #182: EDIT field exits on iddle (with serialIn)

d21d3q, 2018-06-01:
When running [TextField](https://github.com/neu-rah/ArduinoMenu/tree/a422a94a73729a205e22eedc3f761ccca2a00aa7/examples/TextField) example (and using `EDIT` field in other project) I am unable edit anything.
I introduced delay of 1s between `nav.poll()` to observe it. 

```
menu 4.x text field test
4

Main menu
[1]>IP 000.000.000.000
[2] Hex 0000
[3] <Exit
                                               <- sending: *
Main menu
[1]>IP:[0]00.000.000.000
[2] Hex 0000
[3] <Exit
                                               <- not doing anything here
Main menu
[1]>IP 000.000.000.000
[2] Hex 0000
[3] <Exit
                                              <- sending: **
Main menu
[1]>IP:[0]00.000.000.000
[2] Hex 0000
[3] <Exit

Main menu
[1]>IP:>0<00.000.000.000
[2] Hex 0000
[3] <Exit

Main menu
[1]>IP 000.000.000.000
[2] Hex 0000
[3] <Exit
```

So it parses `**` correctly but also exits on next `nav.poll()` with no data provided.

neu-rah, 2018-06-01:
1 - first `*` -> enter field navigation use `+`/`-` to select character position
2 - second `*` -> enter character edit `+`/`-` to select character value
3 - third `*` -> return to field navigation (1)
4 - fourts `*` without changing position -> exit edit mode

you can also use keyboard to enter characters, enter and escape recognized. I use a terminal that sends characters as they are typed (not buffered as arduino IDE monitor), it works better for menu iteration.

done a quick check on my current project using normal text field, so unable to reproduce

neu-rah, 2018-06-01:
on arduino IDE you can choose "No line ending"

want to make it optional on code?

neu-rah, 2018-06-01:
case is here:
https://github.com/neu-rah/ArduinoMenu/blob/master/src/items.cpp#L224

d21d3q, 2018-06-01:
That makes sense. Till now navigating with CR was enough so that I was confused with this behavior.
I don't think that it needs an option in code. I guess it is desired for encoder navigation

neu-rah, 2018-06-01:
probably needs some copy/paste of this info on wiki

neu-rah, 2018-06-02:
done

closed 2018-06-02 by neu-rah