neu-rah/ArduinoMenu #183: Aproach for implementing password protected menu

d21d3q, 2018-06-05:
How would you some kind of lock screen?
I was thinking about using keypad (4x4) can I delegate some keys for navigation, and others as input (and pass them into event handlers?)

d21d3q, 2018-06-05:
quick code lookup and I think that navigation would need to generate some event for passing non navigation keys from keypad. I guess that it would have to take palce somewhere [here](https://github.com/neu-rah/ArduinoMenu/blob/8a58f1bbd515e067fcd91b412a4660e9dfbebd4e/src/nav.cpp#L29)
Or just custom menu item which has own navigation.

Another option is to read directly i main loop keypad (one can read it several times while key is pressed) and manipulate navigation (enable/disable), force displayed node.

neu-rah, 2018-06-05:
yes, keypad hw (as buttons in general) still accessible, it seem desirable to do this outside menu system, no?

neu-rah, 2018-06-05:
there is some code on plugins folder (half abandoned) about custom menu controls
cancelField handles input, and barField handles output

neu-rah, 2020-04-17:
https://github.com/neu-rah/ArduinoMenu/issues/285

neu-rah, 2020-10-20:
we are closer to this as we have `textField` and the question was post again, this time on the gitter...

There is an issue/request about it, but not yet...
I would do it as a separate routine, but one can adapt a field to do it, all it need to print instead is * except when editing, where the current character is shown.. something like this:

class passField:public textField {
public:
  using textField;
  Used printTo(navRoot &root,bool sel,menuOut& out, idx_t idx,idx_t len,idx_t panelNr=0) override;
};
and provide a proper override for `printTo` function

the current implementation is here: https://github.com/neu-rah/ArduinoMenu/blob/master/src/items.cpp#L188

for a quick start i would copy the current implementation and replace https://github.com/neu-rah/ArduinoMenu/blob/master/src/items.cpp#L222 with a simple print of "*", latter i would remove the text cursor, guess we don't need it... and possibly some other things.. as one can also prevent the pass from being peek by just showing edit on the last character, or allow only edit on last and delete