neu-rah/ArduinoMenu #184: User defined graphical menu screen with alternative navigation input

szmolnar, 2018-06-09:
At the moment it is not possible to have a user defined graphical screen on the menu which can display a dynamic content and handle alternative navigation inputs.

Would be nice to have  an output driver implemented that calls an user function.

neu-rah, 2018-06-13:
the current state:

input can be fully customized, see codeCtrl example (https://github.com/neu-rah/ArduinoMenu/tree/master/examples/codeCtrl). One can deal with a custom input and issue menu commands at will.

Output needs a similar counterpart. The idea is to provide a driver that will simply call a user drawing function. I'm still seeking some inspiration from existing menu systems.
The current output system allows customization of menu items but they still have to be confined to the designated area.

About the navigation control, we have a central navigation control code and no provision to customize navigation alternative control methods yet. By navigation control we mean the current selected element and how menu navigation commands are handled and what state changes the do to the navigation state.

TamojitSaha, 2018-10-01:
Closing due to lack of inactivity.

closed 2018-10-01 by TamojitSaha