neu-rah/ArduinoMenu #187: Submenu text does not change with Confirm example

lamqhoang, 2018-06-27:
Hi,
I'm trying with  the Confirm example, but the submenu text is just the same with the confirmExit class text. 
[image]

neu-rah, 2018-06-27:
Hi!  
This is my console output from that example... (https://github.com/neu-rah/ArduinoMenu/blob/master/examples/Confirm/Confirm/Confirm.ino)

```
Main menu
[1] Op1
[2]>Exit

Exit
[1]>Yes
[2] Cancel
Terminating...
done.
```

on your screen you should see message `Terminating...` when selecting `Yes` or typing `1`

are you able to use the menu again after exit?

I've used an arduino mega, what MCU are you using?

neu-rah, 2018-06-27:
sorry, you are talking about the menu text/title, yes its the same and it should not, the purpose was to have customization of title

here's a correction

```c++
//customizing a menu prompt look
class confirmExit:public menu {
public:
  confirmExit(constMEM menuNodeShadow& shadow):menu(shadow) {}
  Used printTo(navRoot &root,bool sel,menuOut& out, idx_t idx,idx_t len,idx_t p) override {
    return idx<0?//idx woll be -1 when printing a menu title or a valid index when printing as option
      menu::printTo(root,sel,out,idx,len,p)://when printing title
      out.printRaw((constText*)F("Exit"),len);//when printing as regular option
  }
};
```

will be included on next push

thank for calling my attention to it

lamqhoang, 2018-06-27:
Thank you for the library, it's really well written. I'm new to this library so I'm not sure about the solution. Your correction works well.
Regards,

neu-rah, 2018-06-27:
you are welcome

feel free to ask on gitter (https://gitter.im/ArduinoMenu/Lobby) if you need any help

please close issue if solved

thanks

closed 2018-06-28 by lamqhoang

lamqhoang, 2018-06-28:
Hi,
Sorry that I forgot to mention. When I test your code. This is the console output :

> menu 4.x custom sub-menu prompt example
> 
> 
> Main menu
> [1]>Op1
> [2] !⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮5⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮
> 
> Exit?
> [1]>Yes
> [2] Cancel
> 

Here is my correction:
```
class confirmExit:public menu {
public:
  confirmExit(constMEM menuNodeShadow& shadow):menu(shadow) {}
  Used printTo(navRoot &root,bool sel,menuOut& out, idx_t idx,idx_t len,idx_t p) override {
    return idx<0?//idx will be -1 when printing a menu title or a valid index when printing as option
      menu::printTo(root,sel,out,idx,len,p)://when printing title
      out.printRaw("Exit",len);//when printing as regular option
  }
};
```

neu-rah, 2018-06-28:
and its a very important detail, glad you mention it
it means "Exit" string need to be in FLASH, please use F("Exit")