neu-rah/ArduinoMenu #189: Incoming uart data streams interferring with the menu system.

Lu0z, 2018-07-15:
Hello. I am that guy who was working on a radio transceiver using your menu system one year ago. I am really sorry i did not give you any news but as i wrote you on facebook i had some life changing situations preventing me to continue working on the project for over a year. The thing is almost complete and i'm running some serious tests on it but i have noticed an issue while working on UART comunications. i hope you could help me one more time. thank you in advance.
 
I was testing the menu system on my UNO  with a uart controlled module. at first what seemed like an EM interference in the board, resulted into an issue generated by the module automated response when a certain command is sent trough a menu function.

The menu funcion which sends the command, holds the first position in the page. and sends the stream as an "exitEvent" function.
Basically when the module, receives the command correctly, it sends back this string to the mcu rx pin: 

+DMOCONNECT: 1<CR><LF>

the menu seems to pick up that "1" in the data stream as a selection command and selects the first menu function again. so i can only change the setting in the field and press the encoder button again to set it. At that point hte routine will send the command to the module once more making it send back the same response so the menu loops back into the first function. It becomes a vicious cycle. I experimented on this thing and i have noticed that using the IDE serial monitor i can control the menu by inputting different numbers and also override value limits on the menu variables. even though this could be useful in some applications i was wondering if there is a way to prevent the menu from picking up data streams incoming from outside into the uart. 
Thank you very much and i hope you did not get angry at me for vanishing in that way.

neu-rah, 2018-07-15:
Hi Luoz, have you seen the cmdCtrl example?
for me it seems better to filter out the stream data and issue menu commands

neu-rah, 2018-07-15:
there is also this https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIo.h#L15

you can set it to false to disable numeric input

```c++
serialIn serial(Serial);
....
serial.numValueInput=false;//on setup
```

neu-rah, 2018-07-27:
Hi!
I've implemented a new navRoot::useAccel option to disable numeric accelerators

example usage, on setup function:
```c++
nav.useAccel=false;
```

closed 2018-07-27 by neu-rah