neu-rah/ArduinoMenu #191: SSD1306 using Adafruit_SSD1306.h

wildview, 2018-08-10:
Nice work on this. This is probably not a issue, but would like to implement your menu system on a project using an SSD1306 Oled display w/ a Teensy v3.2 device. I need to use the Adafruit_SSD1306 and Adafruit_GFX libs since I need to do some graphics work. Does your menu system support using these Adafruit libs versus the SSD1306Ascii code. If so, do you have an example using the Adafruit_SSD1306 display driver. I'm using a Teensy device, so I have lots memory =) 

Regards, R

neu-rah, 2018-08-10:
Hi! thanks for your interest on the lib
The menu library only depends on the graphics lib, so any screen screen that can work with that specific gfx library can in theory be used with the menu
I've been using SSD1306 with u8g2 and it works very nice, using esp8266 and esp32...
we also have this chat room where people use to exchange ideas an some code, you are welcom to join in https://gitter.im/ArduinoMenu/Lobby

neu-rah, 2018-08-18:
need more help?

TamojitSaha, 2018-10-01:
Closing due to lack of inactivity.

closed 2018-10-01 by TamojitSaha