neu-rah/ArduinoMenu #192: Configure Buttons to Navigate in Menu for SSD1306Ascii

TamojitSaha, 2018-08-14:
Please guide me on how to add buttons to navigate through the menu in [this](https://github.com/neu-rah/ArduinoMenu/blob/master/examples/SSD1306Ascii/SSD1306Ascii/SSD1306Ascii.ino) sketch.

neu-rah, 2018-08-14:
Hi!
that sketch uses **serial**
```c++
serialIn serial(Serial);//the serial driver
NAVROOT(nav,mainMenu,MAX_DEPTH,serial,out);//the menu central navigation object
```

you can replace that with the input on this example https://github.com/neu-rah/ArduinoMenu/blob/master/examples/btnNav/btnNav/btnNav.ino

this one uses 3 keys and serial at same time, therefor the usage of an inputs list (would not need if using only keys)

```c++
//define some input pins
#define BTN_SEL 6	// Select button
#define BTN_UP 7 // Up
#define BTN_DOWN 8 // Down

//build a map of keys to menu commands
keyMap joystickBtn_map[]={
 {-BTN_SEL, defaultNavCodes[enterCmd].ch} ,
 {-BTN_UP, defaultNavCodes[upCmd].ch} ,
 {-BTN_DOWN, defaultNavCodes[downCmd].ch}  ,
};
keyIn<3> joystickBtns(joystickBtn_map);//the input driver

serialIn serial(Serial);//another input driver
menuIn* inputsList[]={&joystickBtns,&serial};//a list of inputs
chainStream<2> in(inputsList);//2 is the number of inputs in the list, joining all inputs into one

NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);//using the joined inputs
```
then you can either initialize pin mode making them inputs or
call `joystickBtns.begin()` on setup to do that for you
```c++
joystickBtns.begin();//on setup
```
please let me know if you encounter any problem or if i'm missing something

TamojitSaha, 2018-08-14:
Nice it worked...I am planning to integrate this with SSD1306Ascii.ino
Thanks for the help!

Can this library be ported to ESP8266 and STM32F103C8T6 Board([BluePill Board](https://jeelabs.org/article/1649a/)) ?

neu-rah, 2018-08-14:
This library already works on those MCUs, using the respective Arduino frameworks available for them.
I'm using ESP8266 on my current project and tested on esp32 and blue pill

you are also welcome to join us on chat room https://gitter.im/ArduinoMenu/Lobby

closed 2018-08-14 by neu-rah

TamojitSaha, 2018-08-15:
I was integrating the buttons as you pointed out in the *SSD1306Ascii.ino*. I made the necessary changes. But I want the menu to be independent of Serial Communication. Actually I just need the buttons, OLED and Arduino.

Here is the sketch below
````
#include <Arduino.h>
#define I2C_ADDRESS 0x3C
// #include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#include <menu.h>
#include <menuIO/SSD1306AsciiOut.h>
#include <menuIO/serialIn.h>
using namespace Menu;

SSD1306AsciiWire oled;

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e, navNode& nav, prompt& item) {
  Serial.print("event: ");
  Serial.println(e);
  return proceed;
}

int test = 55;

result action1(eventMask e) {
  Serial.print(e);
  Serial.println(" action1 executed, proceed menu"); Serial.flush();
  return proceed;
}

result action2(eventMask e, navNode& nav, prompt &item) {
  Serial.print(e);
  Serial.println(" action2 executed, quiting menu");
  return quit;
}

int ledCtrl = LOW;

result myLedOn() {
  ledCtrl = HIGH;
  return proceed;
}
result myLedOff() {
  ledCtrl = LOW;
  return proceed;
}

TOGGLE(ledCtrl, setLed, "Led: ", doNothing, noEvent, noStyle //,doExit,enterEvent,noStyle
       , VALUE("On", HIGH, doNothing, noEvent)
       , VALUE("Off", LOW, doNothing, noEvent)
      );

int selTest = 0;
SELECT(selTest, selMenu, "Select", doNothing, noEvent, noStyle
       , VALUE("Zero", 0, doNothing, noEvent)
       , VALUE("One", 1, doNothing, noEvent)
       , VALUE("Two", 2, doNothing, noEvent)
      );

int chooseTest = -1;
CHOOSE(chooseTest, chooseMenu, "Choose", doNothing, noEvent, noStyle
       , VALUE("First", 1, doNothing, noEvent)
       , VALUE("Second", 2, doNothing, noEvent)
       , VALUE("Third", 3, doNothing, noEvent)
       , VALUE("Last", -1, doNothing, noEvent)
      );

//customizing a prompt look!
//by extending the prompt class
class altPrompt: public prompt {
  public:
    altPrompt(constMEM promptShadow& p): prompt(p) {}
    Used printTo(navRoot &root, bool sel, menuOut& out, idx_t idx, idx_t len, idx_t) override {
      return out.printRaw(F("special prompt!"), len);
    }
};

MENU(subMenu, "Sub-Menu", showEvent, anyEvent, noStyle
     , OP("Sub1", showEvent, anyEvent)
     , OP("Sub2", showEvent, anyEvent)
     , OP("Sub3", showEvent, anyEvent)
     , altOP(altPrompt, "", showEvent, anyEvent)
     , EXIT("<Back")
    );

MENU(mainMenu, "Main menu", doNothing, noEvent, wrapStyle
     , OP("Op1", action1, anyEvent)
     , OP("Op2", action2, enterEvent)
     , FIELD(test, "Test", "%", 0, 100, 10, 1, doNothing, noEvent, wrapStyle)
     , SUBMENU(subMenu)
     , SUBMENU(setLed)
     , OP("LED On", myLedOn, enterEvent)
     , OP("LED Off", myLedOff, enterEvent)
     , SUBMENU(selMenu)
     , SUBMENU(chooseMenu)
     , OP("Alert test", doAlert, enterEvent)
     , EXIT("<Back")
    );

#define MAX_DEPTH 2

#define fontW 5
#define fontH 8

//describing a menu output device without macros
//define at least one panel for menu output
const panel panels[] MEMMODE = {{0, 0, 128 / fontW, 64 / fontH}};
navNode* nodes[sizeof(panels) / sizeof(panel)]; //navNodes to store navigation status
panelsList pList(panels, nodes, 1); //a list of panels and nodes
idx_t tops[MAX_DEPTH] = {0, 0}; //store cursor positions for each level
SSD1306AsciiOut outOLED(&oled, tops, pList); //oled output device menu driver
menuOut* constMEM outputs[]  MEMMODE  = {&outOLED}; //list of output devices
outputsList out(outputs, 1); //outputs list

//macro to create navigation control root object (nav) using mainMenu
serialIn serial(Serial);
NAVROOT(nav, mainMenu, MAX_DEPTH, serial, out);

result alert(menuOut& o, idleEvent e) {
  if (e == idling) {
    o.setCursor(0, 0);
    o.print("alert test");
    o.setCursor(0, 1);
    o.print("press [select]");
    o.setCursor(0, 2);
    o.print("to continue...");
  }
  return proceed;
}

result doAlert(eventMask e, prompt &item) {
  nav.idleOn(alert);
  return proceed;
}

//when menu is suspended
result idle(menuOut &o, idleEvent e) {
  o.clear();
  switch (e) {
    case idleStart: o.println("suspending menu!"); break;
    case idling: o.println("suspended..."); break;
    case idleEnd: o.println("resuming menu."); break;
  }
  return proceed;
}

void setup() {
  Serial.begin(115200);
  while (!Serial);
  Serial.println("menu 4.x test"); Serial.flush();
  Wire.begin();
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();
  oled.print("Hello world!");
  nav.idleTask = idle; //point a function to be used when menu is suspended
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay when other tasks are done
}
````

This sketch compiles but the OLED keeps blinking...
I dont understand what the issue is. 😕 Please help me out! 😓

neu-rah, 2018-08-15:
if using that list (not needed for a single input) then you must at least inform the correct list length by changing that 2 into 1
`chainStream<1> in(inputsList)`

TamojitSaha, 2018-08-15:
**UPDATE** 
@neu-rah 
OLED blinking stops after changing to this as you mentioned
````
NAVROOT(nav,mainMenu,MAX_DEPTH,joystickBtns,out)
````
Button navigation is not working if I remove `Serial.begin(115200);`
How to fix this?

neu-rah, 2018-08-15:
the sketch above is your complete code? because i don't see `joystickBtns` there

TamojitSaha, 2018-08-15:
@neu-rah I am really sorry for such a big typo mistake. I just updated the sketch. Please check.

neu-rah, 2018-08-15:
this code compiles, don't know if it runs ok no arduino with ssd1306 right now

```c++
#include <Arduino.h>
#define I2C_ADDRESS 0x3C
// #include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#include <menu.h>
#include <menuIO/SSD1306AsciiOut.h>
#include <menuIO/serialIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/keyMapDef.h>
using namespace Menu;

SSD1306AsciiWire oled;

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e, navNode& nav, prompt& item) {
  Serial.print("event: ");
  Serial.println(e);
  return proceed;
}

int test = 55;

result action1(eventMask e) {
  Serial.print(e);
  Serial.println(" action1 executed, proceed menu"); Serial.flush();
  return proceed;
}

result action2(eventMask e, navNode& nav, prompt &item) {
  Serial.print(e);
  Serial.println(" action2 executed, quiting menu");
  return quit;
}

int ledCtrl = LOW;

result myLedOn() {
  ledCtrl = HIGH;
  return proceed;
}
result myLedOff() {
  ledCtrl = LOW;
  return proceed;
}

TOGGLE(ledCtrl, setLed, "Led: ", doNothing, noEvent, noStyle //,doExit,enterEvent,noStyle
       , VALUE("On", HIGH, doNothing, noEvent)
       , VALUE("Off", LOW, doNothing, noEvent)
      );

int selTest = 0;
SELECT(selTest, selMenu, "Select", doNothing, noEvent, noStyle
       , VALUE("Zero", 0, doNothing, noEvent)
       , VALUE("One", 1, doNothing, noEvent)
       , VALUE("Two", 2, doNothing, noEvent)
      );

int chooseTest = -1;
CHOOSE(chooseTest, chooseMenu, "Choose", doNothing, noEvent, noStyle
       , VALUE("First", 1, doNothing, noEvent)
       , VALUE("Second", 2, doNothing, noEvent)
       , VALUE("Third", 3, doNothing, noEvent)
       , VALUE("Last", -1, doNothing, noEvent)
      );

//customizing a prompt look!
//by extending the prompt class
class altPrompt: public prompt {
  public:
    altPrompt(constMEM promptShadow& p): prompt(p) {}
    Used printTo(navRoot &root, bool sel, menuOut& out, idx_t idx, idx_t len, idx_t) override {
      return out.printRaw(F("special prompt!"), len);
    }
};

MENU(subMenu, "Sub-Menu", showEvent, anyEvent, noStyle
     , OP("Sub1", showEvent, anyEvent)
     , OP("Sub2", showEvent, anyEvent)
     , OP("Sub3", showEvent, anyEvent)
     , altOP(altPrompt, "", showEvent, anyEvent)
     , EXIT("<Back")
    );

MENU(mainMenu, "Main menu", doNothing, noEvent, wrapStyle
     , OP("Op1", action1, anyEvent)
     , OP("Op2", action2, enterEvent)
     , FIELD(test, "Test", "%", 0, 100, 10, 1, doNothing, noEvent, wrapStyle)
     , SUBMENU(subMenu)
     , SUBMENU(setLed)
     , OP("LED On", myLedOn, enterEvent)
     , OP("LED Off", myLedOff, enterEvent)
     , SUBMENU(selMenu)
     , SUBMENU(chooseMenu)
     , OP("Alert test", doAlert, enterEvent)
     , EXIT("<Back")
    );

#define MAX_DEPTH 2

#define fontW 5
#define fontH 8

//describing a menu output device without macros
//define at least one panel for menu output
const panel panels[] MEMMODE = {{0, 0, 128 / fontW, 64 / fontH}};
navNode* nodes[sizeof(panels) / sizeof(panel)]; //navNodes to store navigation status
panelsList pList(panels, nodes, 1); //a list of panels and nodes
idx_t tops[MAX_DEPTH] = {0, 0}; //store cursor positions for each level
SSD1306AsciiOut outOLED(&oled, tops, pList); //oled output device menu driver
menuOut* constMEM outputs[]  MEMMODE  = {&outOLED}; //list of output devices
outputsList out(outputs, 1); //outputs list

//define some input pins
#define BTN_SEL 6	// Select button
#define BTN_UP 7 // Up
#define BTN_DOWN 8 // Down

//build a map of keys to menu commands
keyMap btns_map[]={
 {-BTN_SEL, defaultNavCodes[enterCmd].ch} ,
 {-BTN_UP, defaultNavCodes[upCmd].ch} ,
 {-BTN_DOWN, defaultNavCodes[downCmd].ch}  ,
};
keyIn<3> btns(btns_map);//the input driver

//macro to create navigation control root object (nav) using mainMenu
// serialIn serial(Serial);
NAVROOT(nav, mainMenu, MAX_DEPTH, btns, out);

result alert(menuOut& o, idleEvent e) {
  if (e == idling) {
    o.setCursor(0, 0);
    o.print("alert test");
    o.setCursor(0, 1);
    o.print("press [select]");
    o.setCursor(0, 2);
    o.print("to continue...");
  }
  return proceed;
}

result doAlert(eventMask e, prompt &item) {
  nav.idleOn(alert);
  return proceed;
}

//when menu is suspended
result idle(menuOut &o, idleEvent e) {
  o.clear();
  switch (e) {
    case idleStart: o.println("suspending menu!"); break;
    case idling: o.println("suspended..."); break;
    case idleEnd: o.println("resuming menu."); break;
  }
  return proceed;
}

void setup() {
  // Serial.begin(115200);
  // while (!Serial);
  btns.begin();
  Serial.println("menu 4.x test"); Serial.flush();
  Wire.begin();
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();
  oled.print("Hello world!");
  nav.idleTask = idle; //point a function to be used when menu is suspended
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay when other tasks are done
}
```

neu-rah, 2018-08-15:
and please note the button pins, and replacement for the pins you actually have the buttons in

```c++
#define BTN_SEL 6	// Select button
#define BTN_UP 7 // Up
#define BTN_DOWN 8 // Down
```

TamojitSaha, 2018-08-15:
Running your code and will get back to you shortly.
Sure, I am using the default buttons pin assignement. 😉

TamojitSaha, 2018-08-15:
@neu-rah Your above sketch made my OLED black! No display of menu!

No worries though!
I have modified the sketch to work with Buttons without Serial Communication.
I have tested the code and it works.

**Here is the full modified sketch**
````C

//**********************************************************************
// Sketch: MENU NAVIGATION WITH JUST 3 BUTTONS
//
// This Sketch displays Menu without any Serial Communication and the
// navigation is performed by 3 buttons attached to D6, D7, D8.
//
//  Uses SSD1306Ascii Library(https://github.com/greiman/SSD1306Ascii)
//  by Bill Grieman
//
//  Modifed by Tamojit Saha(https://github.com/TamojitSaha)
//  August 15, 2018
//**********************************************************************
#include <Arduino.h>

#define I2C_ADDRESS 0x3C
// #include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#include <menu.h>
#include <menuIO/SSD1306AsciiOut.h>
//#include <menuIO/SerialIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
using namespace Menu;

SSD1306AsciiWire oled;

#define BTN_SEL 6  // Select button
#define BTN_UP 7 // Up
#define BTN_DOWN 8 // Down

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e, navNode& nav, prompt& item) {
  //Serial.print("event: ");
  //Serial.println(e);
  return proceed;
}

int test = 55;

result action1(eventMask e) {
  //Serial.print(e);
  //Serial.println(" action1 executed, proceed menu");//Serial.flush();
  return proceed;
}

result action2(eventMask e, navNode& nav, prompt &item) {
  //Serial.print(e);
  //Serial.println(" action2 executed, quiting menu");
  return quit;
}

int ledCtrl = LOW;

result myLedOn() {
  ledCtrl = HIGH;
  return proceed;
}
result myLedOff() {
  ledCtrl = LOW;
  return proceed;
}

TOGGLE(ledCtrl, setLed, "Led: ", doNothing, noEvent, noStyle //,doExit,enterEvent,noStyle
       , VALUE("On", HIGH, doNothing, noEvent)
       , VALUE("Off", LOW, doNothing, noEvent)
      );

int selTest = 0;
SELECT(selTest, selMenu, "Select", doNothing, noEvent, noStyle
       , VALUE("Zero", 0, doNothing, noEvent)
       , VALUE("One", 1, doNothing, noEvent)
       , VALUE("Two", 2, doNothing, noEvent)
      );

int chooseTest = -1;
CHOOSE(chooseTest, chooseMenu, "Choose", doNothing, noEvent, noStyle
       , VALUE("First", 1, doNothing, noEvent)
       , VALUE("Second", 2, doNothing, noEvent)
       , VALUE("Third", 3, doNothing, noEvent)
       , VALUE("Last", -1, doNothing, noEvent)
      );

//customizing a prompt look!
//by extending the prompt class
class altPrompt: public prompt {
  public:
    altPrompt(constMEM promptShadow& p): prompt(p) {}
    Used printTo(navRoot &root, bool sel, menuOut& out, idx_t idx, idx_t len, idx_t) override {
      return out.printRaw(F("special prompt!"), len);
    }
};

MENU(subMenu, "Sub-Menu", showEvent, anyEvent, noStyle
     , OP("Sub1", showEvent, anyEvent)
     , OP("Sub2", showEvent, anyEvent)
     , OP("Sub3", showEvent, anyEvent)
     , altOP(altPrompt, "", showEvent, anyEvent)
     , EXIT("<Back")
    );

MENU(mainMenu, "Main menu", doNothing, noEvent, wrapStyle
     , OP("Op1", action1, anyEvent)
     , OP("Op2", action2, enterEvent)
     , FIELD(test, "Test", "%", 0, 100, 10, 1, doNothing, noEvent, wrapStyle)
     , SUBMENU(subMenu)
     , SUBMENU(setLed)
     , OP("LED On", myLedOn, enterEvent)
     , OP("LED Off", myLedOff, enterEvent)
     , SUBMENU(selMenu)
     , SUBMENU(chooseMenu)
     , OP("Alert test", doAlert, enterEvent)
     , EXIT("<Back")
    );

#define MAX_DEPTH 2

#define fontW 5
#define fontH 8

//describing a menu output device without macros
//define at least one panel for menu output
const panel panels[] MEMMODE = {{0, 0, 128 / fontW, 64 / fontH}};
navNode* nodes[sizeof(panels) / sizeof(panel)]; //navNodes to store navigation status
panelsList pList(panels, nodes, 1); //a list of panels and nodes
idx_t tops[MAX_DEPTH] = {0, 0}; //store cursor positions for each level
SSD1306AsciiOut outOLED(&oled, tops, pList); //oled output device menu driver
menuOut* constMEM outputs[]  MEMMODE  = {&outOLED}; //list of output devices
outputsList out(outputs, 1); //outputs list

//build a map of keys to menu commands
keyMap joystickBtn_map[] = {
  { -BTN_SEL, defaultNavCodes[enterCmd].ch} ,
  { -BTN_UP, defaultNavCodes[upCmd].ch} ,
  { -BTN_DOWN, defaultNavCodes[downCmd].ch}  ,
};
keyIn<3> joystickBtns(joystickBtn_map);//the input driver

////SerialIn //Serial(//Serial);//another input driver
//menuIn* inputsList[]={&joystickBtns};//a list of inputs
//chainStream<1> in(inputsList);//2 is the number of inputs in the list, joining all inputs into one

NAVROOT(nav, mainMenu, MAX_DEPTH, joystickBtns, out);

result alert(menuOut& o, idleEvent e) {
  if (e == idling) {
    o.setCursor(0, 0);
    o.print("alert test");
    o.setCursor(0, 1);
    o.print("press [select]");
    o.setCursor(0, 2);
    o.print("to continue...");
  }
  return proceed;
}

result doAlert(eventMask e, prompt &item) {
  nav.idleOn(alert);
  return proceed;
}

//when menu is suspended
result idle(menuOut &o, idleEvent e) {
  o.clear();
  switch (e) {
    case idleStart: o.println("suspending menu!"); break;
    case idling: o.println("suspended..."); break;
    case idleEnd: o.println("resuming menu."); break;
  }
  return proceed;
}

void setup() {
  //Serial.begin(115200);
  //while(!Serial);
  joystickBtns.begin();//
  //Serial.println("menu 4.x test");//Serial.flush();
  Wire.begin();
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();
  oled.print("Hello world!");
  nav.idleTask = idle; //point a function to be used when menu is suspended
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay when other tasks are done
}
````

neu-rah, 2018-08-15:
well glad you found the solution, i've just compiled the code, no assembly to test it ;)

TamojitSaha, 2018-08-15:
I will perform more test and fiddle with the sketch after which, if possible, you can it to *examples* section of this library.

TamojitSaha, 2018-08-15:
I was trying to control brightness of an LED connected to **D6** of  **Arduino UNO**.  I have also attached an event to the `FIELD` section of menu(*refer code below*)

I have a doubts:

- When I am lowering the value by selecting the `Brightness` menu(*refer to the code below*), the LED turns off when the value goes lower than 100%. What I am missing? Please suggest changes.

Here is the full sketch:
````C

//**********************************************************************
// Sketch: MENU NAVIGATION WITH JUST 3 BUTTONS
//
// This Sketch displays Menu without any Serial Communication and the
// navigation is performed by 3 buttons attached to D6, D7, D8.
//
//  Uses SSD1306Ascii Library(https://github.com/greiman/SSD1306Ascii)
//  by Bill Grieman
//
//  Modifed by Tamojit Saha(https://github.com/TamojitSaha)
//  August 15, 2018
//**********************************************************************
#include <Arduino.h>

#define I2C_ADDRESS 0x3C
// #include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#include <menu.h>
#include <menuIO/SSD1306AsciiOut.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
using namespace Menu;

SSD1306AsciiWire oled;

#define BTN_SEL 6  // Select button
#define BTN_UP 7 // Up
#define BTN_DOWN 8 // Down

#define LED_PIN 11  //Define LED Pin to D11

int ledCtrl = LOW;  //Default LED State is LOW

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e, navNode& nav, prompt& item) {
  //Serial.print("event: ");
  //Serial.println(e);
  return proceed;
}

int brightnessValue = 15;

result action1(eventMask e) {
  //Serial.print(e);
  //Serial.println(" action1 executed, proceed menu");//Serial.flush();
  return proceed;
}

result action2(eventMask e, navNode& nav, prompt &item) {
  //Serial.print(e);
  //Serial.println(" action2 executed, quiting menu");
  return quit;
}

result myLedOn() {
  ledCtrl = HIGH;
  digitalWrite(LED_PIN, ledCtrl);
  return proceed;
[... 118 lines cut ...]
  o.clear();
  switch (e) {
    case idleStart: o.println("suspending menu!"); break;
    case idling: o.println("suspended..."); break;
    case idleEnd: o.println("resuming menu."); break;
  }
  return proceed;
}

void setup() {
  joystickBtns.begin();//
  pinMode(LED_PIN, INPUT);
  Wire.begin();
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();
  oled.print("Hello world!");
  nav.idleTask = idle; //point a function to be used when menu is suspended
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay when other tasks are done
}
````

neu-rah, 2018-08-15:
use `map` function, or calculate using floats `round((brightnessValue / 100) * 255);` because when you divide by 100, the result is an integer, therefor if brightnessValue<100 it will be truncated to 0

using map
```c++
analogWrite(LED_PIN, map(brightnessValue,0,100,0,255));
```

or using float
```c++
analogWrite(LED_PIN, 2.55*(float)brightnessValue);
```

TamojitSaha, 2018-08-15:
Thanks it worked like charm.
Posting my full code for other's reference.

````C

//**********************************************************************
// Sketch: MENU NAVIGATION WITH JUST 3 BUTTONS
//
// This Sketch displays Menu without any Serial Communication and the
// navigation is performed by 3 buttons attached to D6, D7, D8.
//
//  Uses SSD1306Ascii Library(https://github.com/greiman/SSD1306Ascii)
//  by Bill Grieman
//
//  Modifed by Tamojit Saha(https://github.com/TamojitSaha)
//  August 15, 2018
//**********************************************************************
#include <Arduino.h>

#define I2C_ADDRESS 0x3C
// #include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#include <menu.h>
#include <menuIO/SSD1306AsciiOut.h>
#include <menuIO/keyIn.h>
//#include <menuIO/chainStream.h>
using namespace Menu;

SSD1306AsciiWire oled;

#define BTN_SEL 6  // Select button
#define BTN_UP 7 // Up
#define BTN_DOWN 8 // Down

#define LED_PIN 11  //Define LED Pin to D11

int ledCtrl = LOW;  //Default LED State is LOW

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e, navNode& nav, prompt& item) {
  //Serial.print("event: ");
  //Serial.println(e);
  return proceed;
}

int brightnessValue = 15;

result action1(eventMask e) {
  //Serial.print(e);
  //Serial.println(" action1 executed, proceed menu");//Serial.flush();
  return proceed;
}

result action2(eventMask e, navNode& nav, prompt &item) {
  //Serial.print(e);
  //Serial.println(" action2 executed, quiting menu");
  return quit;
}

result myLedOn() {
  ledCtrl = HIGH;
  digitalWrite(LED_PIN, ledCtrl);
  return proceed;
}
result myLedOff() {
  ledCtrl = LOW;
  digitalWrite(LED_PIN, ledCtrl);
[... 117 lines cut ...]
  switch (e) {
    case idleStart: o.println("suspending menu!"); break;
    case idling: o.println("suspended..."); break;
    case idleEnd: o.println("resuming menu."); break;
  }
  return proceed;
}

void setup() {
  Serial.begin(9600);
  joystickBtns.begin();//
  pinMode(LED_PIN, INPUT);
  Wire.begin();
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();
  //oled.print("Hello world!");
  nav.idleTask = idle; //point a function to be used when menu is suspended
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay when other tasks are done
}
````

WilixLead, 2018-08-15:
I have a problem with button's input  
```
#include <Arduino.h>
#include <itoa.h>
#include "main.h"

#include <Wire.h>
#include <SSD1306Ascii.h>
#include <SSD1306AsciiWire.h>

#include <menu.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>
#include <menuIO/SSD1306AsciiOut.h>
#include "d-menu.h"
#include "rc.h"

#define I2C_ADDRESS 0x3C

using namespace Menu;

SSD1306AsciiWire oled;

MENU(rcSendMenu,"RC Send",doNothing,noEvent,wrapStyle
  ,OP("Nothing here", doNothing, enterEvent)
  ,EXIT("<Back")
);

TOGGLE(silentMode,silentModeToggle,"SilentMode: ",doNothing,noEvent,noStyle
  ,VALUE("On", true, doNothing, noEvent)
  ,VALUE("Off", false, doNothing, noEvent)
);

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(rcSendMenu)
  ,SUBMENU(silentModeToggle)
  ,OP("Sleep", doNothing, enterEvent)
);

//a keyboard with only one key as the encoder button
//negative pin numbers use internal pull-up, on = low
keyMap encBtn_map[] = {
  {PIN_BTN1, defaultNavCodes[enterCmd].ch},
  {PIN_BTN2, defaultNavCodes[upCmd].ch},
  {PIN_BTN3, defaultNavCodes[downCmd].ch},
  {PIN_BTN4, defaultNavCodes[escCmd].ch},
};
keyIn<4> encButton(encBtn_map);//1 is the number of keys
serialIn serial(Serial);

//input from the encoder + encoder button + serial
menuIn* inputsList[] = {
  &encButton,
  &serial
};
chainStream<2> in(inputsList);

#define MAX_DEPTH 2

constMEM panel panels[] MEMMODE={{0, 0, 128/5, 32/8}};
navNode* nodes[sizeof(panels)/sizeof(panel)];//navNodes to store navigation status
panelsList pList(panels,nodes,1);//a list of panels and nodes
idx_t tops[MAX_DEPTH]={0,0};//store cursor positions for each level
SSD1306AsciiOut outOLED(&oled,tops,pList);//oled output device menu driver
serialOut sOut(Serial, tops);

menuOut* outputs[]={&outOLED, &sOut};//list of output devices
outputsList out(outputs,2);//outputs list

NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);

void menuSetup() {
  Wire.begin();
  Wire.setClock(400000L);
  
  oled.begin(&Adafruit128x32, I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();
  encButton.begin();
}

void menuLoop() {
  nav.poll();
}
```

TamojitSaha, 2018-08-15:
> //a keyboard with only one key as the encoder button
//negative pin numbers use internal pull-up, on = low
keyMap encBtn_map[] = {
  {PIN_BTN1, defaultNavCodes[enterCmd].ch},
  {PIN_BTN2, defaultNavCodes[upCmd].ch},
  {PIN_BTN3, defaultNavCodes[downCmd].ch},
  {PIN_BTN4, defaultNavCodes[escCmd].ch},
};

Where are the buttons defined??

WilixLead, 2018-08-15:
In main.h file, it look's like this:
```
#define PIN_BTN1          PB12
#define PIN_BTN2          PB13
#define PIN_BTN3          PB14
#define PIN_BTN4          PA0
```

TamojitSaha, 2018-08-15:
Here is the modified code that works with buttons. I have commented some headers so as to my setup. Change the necessary according to you.

````C
#include <Arduino.h>
//#include <itoa.h>
//#include "main.h"
#define MAX_DEPTH 2

#define fontW 5
#define fontH 8

#include <Wire.h>
#include <SSD1306Ascii.h>
#include <SSD1306AsciiWire.h>

#include <menu.h>
#include <menuIO/keyIn.h>
//#include <menuIO/chainStream.h>
//#include <menuIO/serialOut.h>
//#include <menuIO/serialIn.h>
#include <menuIO/SSD1306AsciiOut.h>
//#include "d-menu.h"
//#include "rc.h"

#define I2C_ADDRESS 0x3C

using namespace Menu;

#define PIN_BTN1 6 //Select Button
#define PIN_BTN2 7 //Up Button
#define PIN_BTN3 8 //Down Button
#define PIN_BTN4 5 //ESC Button

int silentMode = false;

SSD1306AsciiWire oled;

MENU(rcSendMenu,"RC Send",doNothing,noEvent,wrapStyle
  ,OP("Nothing here", doNothing, enterEvent)
  ,EXIT("<Back")
);

TOGGLE(silentMode,silentModeToggle,"SilentMode: ",doNothing,noEvent,noStyle
  ,VALUE("On", true, doNothing, noEvent)
  ,VALUE("Off", false, doNothing, noEvent)
);

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(rcSendMenu)
  ,SUBMENU(silentModeToggle)
  ,OP("Sleep", doNothing, enterEvent)
);

constMEM panel panels[] MEMMODE={{0, 0, 128/5, 32/8}};
navNode* nodes[sizeof(panels) / sizeof(panel)]; //navNodes to store navigation status
panelsList pList(panels, nodes, 1); //a list of panels and nodes
idx_t tops[MAX_DEPTH] = {0, 0}; //store cursor positions for each level
SSD1306AsciiOut outOLED(&oled, tops, pList); //oled output device menu driver
menuOut* constMEM outputs[]  MEMMODE  = {&outOLED}; //list of output devices
outputsList out(outputs, 1); //outputs 

//a keyboard with only one key as the encoder button
//negative pin numbers use internal pull-up, on = low
keyMap encBtn_map[] = {
  {-PIN_BTN1, defaultNavCodes[enterCmd].ch},
  {-PIN_BTN2, defaultNavCodes[upCmd].ch},
  {-PIN_BTN3, defaultNavCodes[downCmd].ch},
  {-PIN_BTN4, defaultNavCodes[escCmd].ch},
};
keyIn<4> encButton(encBtn_map);//1 is the number of keys
//serialIn serial(Serial);

//input from the encoder + encoder button + serial
NAVROOT(nav, mainMenu, MAX_DEPTH, encButton, out);

void setup() {
  Wire.begin();
  Wire.setClock(400000L);
  
  oled.begin(&Adafruit128x32, I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();
  encButton.begin();
}

void loop() {
  nav.poll();
}
````

TamojitSaha, 2018-08-15:
````C
//#include <menuIO/chainStream.h>
//#include <menuIO/serialOut.h>
//#include <menuIO/serialIn.h>
````
This headers are not needed if you don't plan to use `Serial`

WilixLead, 2018-08-15:
I found solution. It is related to keyIn code inside. I use stm32 and pull-upped inputs. So for normal click handling I need INPUT_PULLDOWN, but in keyIn code if key is mode then 0 used INPUT_PULLUP and if less then 0 it use INPUT.