neu-rah/ArduinoMenu #193: Font overlapping in SSD1306Ascii Library

TamojitSaha, 2018-08-17:
I was creating menu with fonts like `Callibri11` or `X11fixed7x14` or `Verdana12` other than default `System5x7`. The fonts are getting overlapped while display in an *128x64 SSD1306 OLED*. I couldnt manage to correct it.
[Here](https://drive.google.com/file/d/1SiYQQlCTmsZSRGQoMCv0UMMOamlUgssc/view?usp=sharing) I have attached how the display looks :

**Code Snippets:**
````C
//Menu Display Section
#define fontW 8
#define fontH 12

//describing a menu output device without macros
//define at least one panel for menu output
const panel panels[] MEMMODE = {{0, 0, 128 / fontW, 64 / fontH}};
navNode* nodes[sizeof(panels) / sizeof(panel)]; //navNodes to store navigation status
panelsList pList(panels, nodes, 1); //a list of panels and nodes
idx_t tops[MAX_DEPTH] = {0, 0}; //store cursor positions for each level
SSD1306AsciiOut outOLED(&oled, tops, pList); //oled output device menu driver
menuOut* constMEM outputs[]  MEMMODE  = {&outOLED}; //list of output devices
outputsList out(outputs, 1); //outputs list

//void setup()
void setup() {
  Serial.begin(9600);
  joystickBtns.begin();//
  pinMode(LED_PIN, INPUT);
  Wire.begin();
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
  oled.setFont(Verdana12);
  oled.clear();
  //oled.print("Hello world!");
  nav.idleTask = idle;
  //point a function to be used when menu is suspended
}

````

neu-rah, 2018-08-17:
by checking the driver https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIO/SSD1306AsciiOut.h#L15 i can see its a text driver, so it uses text coordinates to print (or so I expect)  
on that device how do you produce 2 non-overlapping lines?

TamojitSaha, 2018-08-17:
@neu-rah did you see tha attached picture? I cannot understand why the output is such?

neu-rah, 2018-08-17:
yes, i saw the picture, it happens on graphics displays when line height is not properly set, but text devices work on line numbers, so what is the result of printing with that font to line 0 and 1? i gota be sure this is ok at screen library level

TamojitSaha, 2018-08-17:
I did not understand what you meant by text devices. 

I used SSD1306 I2C OLED as the graphics display for the library with *SSD1306Ascii* example sketch. I just changed the default font to either of the font I listed above and the result was peculiar. 
As you said this the problem of line-height, then is there any way I can change any parameters to display the characters correctly?

neu-rah, 2018-08-18:
that **Ascii** on the driver name means that it uses text to be a fast driver and the menu driver for SSD1306Ascii was made as a text device, meaning that the menu print first option at line 0 and second option on line 1. But it looks that the SSD1306ASCII driver needs some more setup to work with a larger font. I wonder what the result would be by doing this on setup.

```c++
oled.setCursor(0,0);
oled.write("line 0");
oled..setCursor(0,1);
oled.write("line 1");
```

please excuse inaccuracies of above code... i've just looked at the ssd1306ascii docs and of course not compiling this. but i suspect that it will produce and overlapping too.
By default the ssd1306ascii uses 8 pixels character boxes, and it looks that changing the font is not sufficient, and as so that set2x() api function makes sense... if so you just have to add it to setup and have everything in place (I hope)

TamojitSaha, 2018-08-18:
The *SSD1306Ascii* alone just works fine with other fonts when used individually. The issue is with the *ArduinoMenu* library. It does not print correctly with bigger fonts than *5x7*

TamojitSaha, 2018-08-18:
Can you please point me out where *SSD1306Ascii* library is used internally to print out the menu??

neu-rah, 2018-08-18:
of course  
`menuIO/SSD1306AsciiOut.h` and `menuIO/SSD1306AsciiOut.cpp`
you will probably also have problems if using a non mono-metric (all char with same width) font

because the driver is text based it will erase the line by drawing spaces, however spaces on non mono-metric fonts use to be quite narrow and will not cover all the line

I'm trying to rework the driver into a gfx driver instead... but not ready yet and i'm not sure if its worth, because even 12px height font will have to be placed into 16px height lines, and right now i suspect that it has to be 17px (one px for line spacing?). If so the device will only have 3 complete lines...

neu-rah, 2018-08-19:
hi!
turned SSD1306AsciiOut into a gfxOut derived and updated the example on this commit https://github.com/neu-rah/ArduinoMenu/tree/696c1e599dda927fc2f6703d80cf8686fb58c045

menu working ok with 8 and 12 px height font, provided that the correct scaling is entered at constructor

this device still uses text cursor `>`, until i have a text color inversion...

TamojitSaha, 2018-08-19:
Wow, you are a genius...I will be testing this by today. Thanks man!!!

TamojitSaha, 2018-08-19:
@neu-rah Updated my comment. Please check

neu-rah, 2018-08-19:
you are welcome :D

neu-rah, 2018-08-19:
this can also be

```c++
  #define fontW 8
  #define fontH 16
  SSD1306AsciiOut outOLED(&oled,tops,pList,fontW,fontH>>3);
```

closed 2018-08-19 by neu-rah

dhysuiej, 2023-03-14:
I think I'm having the [same problem](https://github.com/olikraus/u8g2/issues/2133) but with the driver u8x8. Could someone please point me in the right direction to figure out a workaround.

[image]