neu-rah/ArduinoMenu #198: ESP8266 - Nodemcu compile problem

gmgunderground, 2018-08-19:
Hi, unfortunately (this library seams to be very good) I'm not able to compile a standard example for my board, I'using Nodemcu with PlatformIO with latest toolchane and last IDE. The problem is in this code

MENU_OUTPUTS(out,MAX_DEPTH
  ,SERIAL_OUT(Serial)
  ,NONE//must have 2 items at least
);

> Compiling .pioenvs\nodemcuv2\src\main.cpp.o
In file included from lib\MENU/menuBase.h:31:0,
from lib\MENU/menuDefs.h:11,
from lib\MENU/menu.h:15,
from src\main.cpp:13:
lib\MENU/macros.h:273:111: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
constMEM Menu::menuFieldShadow<typeof(target)>& _fieldShadow##cnt=*(Menu::menuFieldShadow<typeof(target)>*)&fieldShadowRaw##cnt;\
^
lib\MENU/macros.h:51:17: note: in expansion of macro 'DECL_FIELD_'
#define DECL(x) DECL_##x
^
lib\MENU/baseMacros.h:14:3: note: in expansion of macro 'DECL'
what(x)\
^
lib\MENU/baseMacros.h:7:35: note: in expansion of macro 'FOR_EACH_3'
#define CONCATENATE2(arg1, arg2)  arg1##arg2
^
lib\MENU/baseMacros.h:109:32: note: in expansion of macro 'FOR_EACH_'
#define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)
^
lib\MENU/macros.h:170:3: note: in expansion of macro 'FOR_EACH'
FOR_EACH(DECL,__VA_ARGS__)\
^
lib\MENU/macros.h:167:42: note: in expansion of macro 'altMENU'
#define MENU(id,text,aFn,mask,style,...) altMENU(Menu::menu,id,text,aFn,mask,style,(Menu::_menuData|Menu::_canNav),__VA_ARGS__)
^
src\main.cpp:26:1: note: in expansion of macro 'MENU'
MENU(mainMenu, "Blink menu", Menu::doNothing, Menu::noEvent, Menu::wrapStyle
^
lib\MENU/macros.h:273:111: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
constMEM Menu::menuFieldShadow<typeof(target)>& _fieldShadow##cnt=*(Menu::menuFieldShadow<typeof(target)>*)&fieldShadowRaw##cnt;\
^
lib\MENU/macros.h:51:17: note: in expansion of macro 'DECL_FIELD_'
#define DECL(x) DECL_##x
^
lib\MENU/baseMacros.h:11:3: note: in expansion of macro 'DECL'
what(x)\
^
lib\MENU/baseMacros.h:15:3: note: in expansion of macro 'FOR_EACH_2'
FOR_EACH_2(what, __VA_ARGS__)
^
lib\MENU/baseMacros.h:7:35: note: in expansion of macro 'FOR_EACH_3'
#define CONCATENATE2(arg1, arg2)  arg1##arg2
^
lib\MENU/baseMacros.h:109:32: note: in expansion of macro 'FOR_EACH_'
#define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)
^
lib\MENU/macros.h:170:3: note: in expansion of macro 'FOR_EACH'
FOR_EACH(DECL,__VA_ARGS__)\
^
lib\MENU/macros.h:167:42: note: in expansion of macro 'altMENU'
#define MENU(id,text,aFn,mask,style,...) altMENU(Menu::menu,id,text,aFn,mask,style,(Menu::_menuData|Menu::_canNav),__VA_ARGS__)
^
src\main.cpp:26:1: note: in expansion of macro 'MENU'
MENU(mainMenu, "Blink menu", Menu::doNothing, Menu::noEvent, Menu::wrapStyle
^
lib\MENU/macros.h:233:69: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
constMEM Menu::promptShadow& opShadow##cnt=*(Menu::promptShadow*)&opShadowRaw##cnt;\
^
lib\MENU/macros.h:51:17: note: in expansion of macro 'DECL_EXIT_'
#define DECL(x) DECL_##x
^
lib\MENU/baseMacros.h:9:34: note: in expansion of macro 'DECL'
#define FOR_EACH_1(what, x, ...) what(x)
^
lib\MENU/baseMacros.h:12:3: note: in expansion of macro 'FOR_EACH_1'
FOR_EACH_1(what,  __VA_ARGS__)
^
lib\MENU/baseMacros.h:15:3: note: in expansion of macro 'FOR_EACH_2'
FOR_EACH_2(what, __VA_ARGS__)
^
lib\MENU/baseMacros.h:7:35: note: in expansion of macro 'FOR_EACH_3'
#define CONCATENATE2(arg1, arg2)  arg1##arg2
^
lib\MENU/baseMacros.h:109:32: note: in expansion of macro 'FOR_EACH_'
#define FOR_EACH(what, x, ...) FOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, x, __VA_ARGS__)
^
lib\MENU/macros.h:170:3: note: in expansion of macro 'FOR_EACH'
FOR_EACH(DECL,__VA_ARGS__)\
^
lib\MENU/macros.h:167:42: note: in expansion of macro 'altMENU'
#define MENU(id,text,aFn,mask,style,...) altMENU(Menu::menu,id,text,aFn,mask,style,(Menu::_menuData|Menu::_canNav),__VA_ARGS__)
^
src\main.cpp:26:1: note: in expansion of macro 'MENU'
MENU(mainMenu, "Blink menu", Menu::doNothing, Menu::noEvent, Menu::wrapStyle
^
In file included from lib\MENU/menuBase.h:31:0,
from lib\MENU/menuDefs.h:11,
from lib\MENU/menu.h:15,
from src\main.cpp:13:
lib\MENU/macros.h:77:27: error: expected constructor, destructor, or type conversion before '(' token
#define SERIAL_OUT(...) ON(SERIAL_OUT,__COUNTER__,__VA_ARGS__)
^
lib\MENU/macros.h:75:31: note: in definition of macro 'CALL__'
#define CALL__(a,b,...) a##_##b(__VA_ARGS__)
^
lib\MENU/macros.h:69:19: note: in expansion of macro 'CALL_'
#define CALL(a,b) CALL_(HEAD_##a,HEAD_##b,TAIL_##a,TAIL_##b)
^
lib\MENU/baseMacros.h:58:3: note: in expansion of macro 'CALL'
what(params,x)\
^
lib\MENU/baseMacros.h:7:35: note: in expansion of macro 'XFOR_EACH_2'
#define CONCATENATE2(arg1, arg2)  arg1##arg2
^
lib\MENU/baseMacros.h:112:42: note: in expansion of macro 'XFOR_EACH_'
#define XFOR_EACH(what, params , x, ...) XFOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, params, x, __VA_ARGS__)
^
lib\MENU/macros.h:161:2: note: in expansion of macro 'XFOR_EACH'
XFOR_EACH(CALL,WITH(VAR,id,maxDepth),__VA_ARGS__)\
^
src\main.cpp:35:1: note: in expansion of macro 'MENU_OUTPUTS'
MENU_OUTPUTS(out,MAX_DEPTH
^
src\main.cpp:36:4: note: in expansion of macro 'SERIAL_OUT'
,SERIAL_OUT(Serial)
^
src\main.cpp:35:14: error: 'out_outPtrs' was not declared in this scope
MENU_OUTPUTS(out,MAX_DEPTH
^
lib\MENU/macros.h:165:22: note: in definition of macro 'MENU_OUTPUTS'
Menu::outputsList id(id##_outPtrs,sizeof(id##_outPtrs)/sizeof(Menu::menuOut*));
^
src\main.cpp:35:14: error: 'out_outPtrs' was not declared in this scope
MENU_OUTPUTS(out,MAX_DEPTH
^
lib\MENU/macros.h:165:42: note: in definition of macro 'MENU_OUTPUTS'
Menu::outputsList id(id##_outPtrs,sizeof(id##_outPtrs)/sizeof(Menu::menuOut*));
^

neu-rah, 2018-08-19:
i was able to compile to wemos mini D1 using platformio, check updates and report please

just had to disable your "board.h" include
```c++
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <Wire.h>
// #include "board.h"

#include <menu.h>
#include <menuIO/serialOut.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialIn.h>

using namespace Menu;

#define LEDPIN LED_BUILTIN
#define MAX_DEPTH 1

int timeOn=10;
int timeOff=90;

MENU(mainMenu, "Blink menu", Menu::doNothing, Menu::noEvent, Menu::wrapStyle
  ,FIELD(timeOn,"On","ms",0,100,10,1, Menu::doNothing, Menu::noEvent, Menu::noStyle)
  ,FIELD(timeOff,"Off","ms",0,100,10,1,Menu::doNothing, Menu::noEvent, Menu::noStyle)
  ,EXIT("<Back")
);

serialIn serial(Serial);
MENU_INPUTS(in,&serial);

MENU_OUTPUTS(out,MAX_DEPTH
  ,SERIAL_OUT(Serial)
  ,NONE//must have 2 items at least
);

NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);

// put your setup code here, to run once:
void setup()
{
    Serial.begin(9600);
    Wire.begin(SDA, SCL);
}

void loop()
{
    nav.poll();
    //deviceCfgMEnager();
}```

gmgunderground, 2018-08-19:
Hi, you are right in board.h there is a ON define, now all is compiled but there is a problem with linking

Linking .pioenvs\nodemcuv2\firmware.elf
.pioenvs\nodemcuv2\src\main.cpp.o:(.text.startup._GLOBAL__sub_I_timeOn+0x48): undefined reference to `Menu::default_serial_panel_list'
collect2.exe: error: ld returned 1 exit status

neu-rah, 2018-08-19:
uploaded to esp8266 ok, i can navigate the menu over serial...
i'm using latest ArduinoMenu varsion v4.13.x
platformio updated

does it persist?

gmgunderground, 2018-08-19:
Now It's working, I don't know why, under PlatformIO, if I copy only SRC content in Lib/myfolder/ArduinoMenuSrcFiles doesn't work, but if I copy all repository content in Lib/myFolder/ArduinoMenuRoot works correctly. Now the issue is solved, It seams to be a very good library, I will play now... thanks for your support

closed 2018-08-20 by neu-rah