neu-rah/ArduinoMenu #202: Customize numeric field edit cursors

neu-rah, 2018-08-27:
as asked by @lamqhoang at gitter channel

> Can I set the FIELD menu indicator instead of the default ":" and ">"

this is quite easy to do and no impact on library size, at this moment the fieldBase class is printing the cursors hard coded, they can however be turned into static fieldBase members and possibly using strings instead of characters (to allow UTF8 stuff)

xgallom, 2019-03-29:
Hi neu-rah, I've implemented it the way you suggested it:

```
diff --git a/src/items.cpp b/src/items.cpp
index 4abb48e..c5c629b 100644
--- a/src/items.cpp
+++ b/src/items.cpp
@@ -1,6 +1,10 @@
 #include "menuDefs.h"
 using namespace Menu;
 
+const char
+  *fieldBase::TunningCursor = ">",
+  *fieldBase::NonTunningCursor = ":";
+
 bool prompt::hasTitle(navNode& nav) const {return (nav.target->has(showTitle)||(nav.root->showTitle&&!nav.target->has(noTitle)));}
 
 idx_t prompt::printRaw(menuOut& out,idx_t len) const {
@@ -389,7 +393,7 @@ Used fieldBase::printTo(navRoot &root,bool sel,menuOut& out, idx_t idx,idx_t len
     //TODO: remove cursor when printing asPad menus
     // if(!root.node().has(_asPad))//do notdraw edit cursors for pad menu items?
     //this is not really working!
-    out.print((root.navFocus==this&&sel)?(tunning?'>':':'):' ');
+    out.print((root.navFocus==this&&sel) ? (tunning ? TunningCursor : NonTunningCursor) : " ");
     l++;
     if (l<len) {
       #ifdef MENU_FMT_WRAPS
diff --git a/src/items.h b/src/items.h
index a08617b..e114bc0 100644
--- a/src/items.h
+++ b/src/items.h
@@ -172,6 +172,10 @@
         #ifdef MENU_ASYNC
           const char* typeName() const override;
         #endif
+
+        static const char
+          *TunningCursor,
+          *NonTunningCursor;
     };
     //--------------------------------------------------------------------------
     template<typename T>

```

neu-rah, 2019-03-29:
Hi xgallom
excellent, thank you!
wanna do a pull request?

xgallom, 2019-03-29:
For sure!