neu-rah/ArduinoMenu #204: ISO C++ forbids converting a string error

array81, 2018-09-11:
I'm trying to compile LiquidCrystal demo on my Aurduino Nano (ch340, a chinese version). I have installed your PCINT but when try to upload the code I get many error like:
`ISO C++ forbids converting a string constant to 'char*'`

neu-rah, 2018-09-11:
Hi! I had an error compiling that example for nano
solved by adding `#include <Wire.h>` to the example... in spit of it not using i2c...

the error that I've got is different from that one, so please let me know if any error persist

pushing fix

array81, 2018-09-12:
This NOT solve the problem.
I have tested this examples:
examples\LCDs\LiquidCrystal\LiquidCrystal
examples\LCDs\Malpartida\I2C\I2C

one of this demo juct include Wire.h

neu-rah, 2018-09-12:
what IDE are you using?

I have warning when using arduino IDE
`warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 char* constMEM`
not errors

neu-rah, 2018-09-13:
you are correct, the error is from PCINT, not showing on my machine because it was one of the changes to PCINT about to be released, pushing changes now.

thank for your feedback, its greatly appreciated

array81, 2018-09-13:
Thanks, now can I compile and upload.

Do you have a sketch about I2C example? I'm try to create it but button doesn't work and I'm not an expert.

neu-rah, 2018-09-14:
nice! you are welcome

I2C sketch... not really, I2C is quite easy to connect and web is full of sketches for many boards, but guess your problem is not there.
The buttons, just connect to the pin and to ground, sketch is using internal pull-ups resistors on input pin.

on **i2c.ino**
`#define encBtn 4` encoder button should be on this pin (https://github.com/neu-rah/ArduinoMenu/blob/master/examples/LCDs/Malpartida/I2C/I2C/I2C.ino#L34)

setup initializes the pin here https://github.com/neu-rah/ArduinoMenu/blob/master/examples/LCDs/Malpartida/I2C/I2C/I2C.ino#L195

if you are using an encoder that already has its own pull-ups resistor, you can disable the pull up there

the buttons are active high, but that can be changed also, let me know how it went...

array81, 2018-09-14:
For understand:
PIN A3 -> LED;
PIN D2 -> button;
PIN D3 -> button;
Encoder button is a knob on PIN A4 or a button on PIN D4?

All button without resistors because you code use internal pull-ups resistors.

neu-rah, 2018-09-14:
on that sketch:
encA 2 ->its one pin of the encoder
encB 3 -> its the other pin of the encoder
encBtn 4 -> its a button, can be the encoder button if it has one
and yea, LED on A3

encoder is a quadrature encoder with optional button, quadrature has 2 pins

neu-rah, 2018-09-14:
it can be a raw encoder like this one

https://www.ebay.com/itm/2pcs-Rotary-encoder-with-switch-EC11-Audio-digital-potentiometer-20mm-handle/311425031990?hash=item488260df36:g:GnMAAOSwv-NWUrCs

or with hardware debounce and pull-ups
https://www.ebay.com/itm/Rotary-Encoder-Module-Brick-Sensor-Development-Board-Potentiometer-5V-/332775079350?hash=item4d7af0bdb6

array81, 2018-09-14:
Thanks for your reply. I did not know this encoder. I just have order it. I'll let you know how it will go.