neu-rah/ArduinoMenu #206: Support for i2c keypads

vladkozlov69, 2018-09-19:
It would be nice to have support for i2c keypads (example of hardware implementation is https://playground.arduino.cc/Main/I2CPortExpanderAndKeypads). It will allow users to use 2 pins instead of 8 and for big projects it could be very important.

neu-rah, 2018-09-19:
I think I have one of those around, will take a look at it... unless someone wants to do it

vladkozlov69, 2018-09-20:
Hi Rui,

I tried to make it by myself... I took a library for I2C Keypad from https://github.com/whatnick/i2ckeypad and added I2CkeypadIn.h with the following code:

```
#ifndef RSITE_I2CKEYPAD_DEF_H
  #define RSITE_I2CKEYPAD_DEF_H
  #include "../menuDefs.h"

  #include <i2ckeypad.h>
  namespace Menu {

    class I2CkeypadIn:public menuIn {
    public:
      char key=0;
      i2ckeypad * in;
      I2CkeypadIn(i2ckeypad * in):in(in) {}
      int available(void) {
		  return peek()!=0;
	  }
      int peek(void) { 
		  return key != 0 ? key : (key=in->get_key()) ?key : 0;
	  }
      int read() {
        if (key != 0) {
          char k=key;
          key=0;
          return k;
        }
        return available()?read():0;
      }
      void flush() {}
      size_t write(uint8_t v) {return 0;}
    };
  };
#endif
```

So far it works as expected...

neu-rah, 2018-09-20:
excellent, if you fell like sharing it a pull request will be welcome

vladkozlov69, 2018-09-20:
Ok, will do

closed 2018-09-20 by neu-rah