neu-rah/ArduinoMenu #209: Anyway to use idle to display changing values

Ken-Fitz, 2018-09-25:
Is there a way to use the idle task to display changing values, so that when the menu times out, or you go back it displays a list of values that are changing?

neu-rah, 2018-09-26:
Hi!
Yes, the menu does not store data, instead it uses references to data already existing on your sketch. So you already have access to the data, but guess you already know that. Only putting this here for future reference.
To signal the menu system that the idle screen need to be redraw to reflect changes you can set `navRoot::idleChanged` to true indide your idle handler

neu-rah, 2018-09-26:
Just to clarify, you must be wondering why idle isn't continuously called with `idling`, and if so why it exists...
well, some devices require a constant full redraw and that device redraw can take multiple menu draws (raster devices).
Being a minimalist (not on size) system we will only call it when the devices so requires and for the number of times needed.

Ken-Fitz, 2018-09-27:
Works wonderfully.

closed 2018-09-27 by Ken-Fitz