neu-rah/ArduinoMenu #210: Encoder button accuracy on NodeMCU v2

array81, 2018-09-28:
I have a low accuracy for encoder button when use NodeMCU. On Arduino Nano it work well but on NodeMCU (same configuration: LCD, encoder) when rotate the encoder sometimes I get the right result for example the cursor is moved to next o before item but sometimes the cursor is moved to many items.
Click button works well.

vladkozlov69, 2018-09-28:
I'm observing the same on ESP32. I thought it was my encoder, because simple encoder test sketch shows increment or decrement by 2 instead of one. I'm waiting now for the several encoder models from Ali to test it.

neu-rah, 2018-09-28:
what encoder driver is in use?
menuIO/clickEncoderIn.h
menuIO/encoderIn.h

i'm using clickEncoderIn on esp8266
also be sure to remove main loop delay used on most examples...

array81, 2018-09-28:
I use (un demo IC2) menuIO/encoderIn.h (with menuIO/clickEncoderIn.h I get error about ClickEncoder.h not found), and I have remove delay(100); from demo.

neu-rah, 2018-09-29:
thanks for the feedback, i have an esp32 around will give it a try

neu-rah, 2018-09-30:
@array81 missing file is from encoder driver https://github.com/0xPIT/encoder , must install it
you will also have to use some esp timer, the example is using timerOne (i think) and it was not compatible with esp last time i checked... please let me know if you need esp timer setup, i have some working code

array81, 2018-10-01:
If you send me your code, you'll do me a favor. timerOne doesn't seem to work on NodeMCU.

neu-rah, 2018-10-01:
using clickEncoder (0XPIT encoder) with esp

```c++
extern "C" {//esp8266 stuff
  #include "user_interface.h"
}

#include <Arduino.h>
#include <ESP8266WiFi.h>

#include <Wire.h>//oled+? bus
#include <U8g2lib.h>//oled driver
// #include <ClickEncoder.h>//from https://github.com/0xPIT/encoder

#define I2C_ADDRESS 0x3C
// #include "SSD1306Ascii.h"
// #include "SSD1306AsciiWire.h"
#include <menu.h>
// #include <menuIO/SSD1306AsciiOut.h>

#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/u8g2Out.h>

using namespace Menu;

//monocrome colors
const colorDef<uint8_t> colors[]={
  {{0,0},{0,1,1}},//bgColor
  {{1,1},{1,0,0}},//fgColor
  {{1,1},{1,0,0}},//valColor
  {{1,1},{1,0,0}},//unitColor
  {{0,1},{0,0,1}},//cursorColor
  {{1,1},{1,0,0}},//titleColor
};
#define menuFont u8g2_font_7x13_mf
constexpr int fontX=7;
constexpr int fontY=16;

// SSD1306AsciiWire oled;
#define OLED_ROT U8G2_R2
constexpr int OLED_SDA=0;
constexpr int OLED_SDC=2;
U8G2_SSD1306_128X64_NONAME_F_HW_I2C oled(OLED_ROT, U8X8_PIN_NONE, OLED_SDC, OLED_SDA);

#define encA 5
#define encB 4
#define encBtn 10
#define encSteps 4

ClickEncoder clickEncoder = ClickEncoder(encA,encB,encBtn,encSteps);
ClickEncoderStream encStream(clickEncoder,1);

//----------------------------------------------------------

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e,navNode& nav,prompt& item) {
  Serial.print("event: ");
  Serial.println(e);
  return proceed;
}

int test=55;

result action1(eventMask e) {
  Serial.print(e);
  Serial.println(" action1 executed, proceed menu");
[... 156 lines cut ...]
  // if (lastService + 1000 < micros()) {
  //   lastService = micros();
  //   clickEncoder.service();
  //
  // }

  //--------------------Menu:------------------------
  constexpr int menuFPS=1000/30;
  static unsigned long lastMenuFrame=-menuFPS;
  unsigned long now=millis();
  if (now-lastMenuFrame>=menuFPS) {
    lastMenuFrame=millis();
    // Serial.write('.');
    Serial.print(digitalRead(encA));
    Serial.print(digitalRead(encB));
    Serial.println(digitalRead(encBtn));
    nav.doInput();
    if (nav.changed(0)||(nav.sleepTask&&nav.idleChanged)) {//gfx device [0] changed?
      oled.setFont(menuFont);
      oled.firstPage();
      do nav.doOutput(); while(oled.nextPage());
    }
  }
}
```

array81, 2018-10-01:
OK. I have comment this lines on ClickEncoder.h
`//#include <avr/io.h>
//#include <avr/interrupt.h>
//#include <avr/pgmspace.h>`

I can compile and upload the code and encoder seems work. I get this message after upload the code:
`warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed`

array81, 2018-10-02:
I don't know if can help you but this demo seems works well with NodeMCU: https://arduino.stackexchange.com/questions/30012/how-to-use-a-rotary-encoder-on-nodemcu-with-arduino-code

array81, 2018-10-09:
Sorry for close it. I have wrong.

I'd like know if there are new

neu-rah, 2018-10-09:
sorry, not yet, been away
hope to return to this soon
thanks

neu-rah, 2018-10-30:
HI, no, not yet (shame :disappointed: )

array81, 2018-11-09:
Some information about problem. I hope this can help you to solve it.
1. if I rotate clockwise I have the result expected: the cursor is move of one item downward for any encoder step;
2. if I rotate counterclockwise slowly the cursor is not move. But if I rotate counterclockwise quickly the cursor is of one item upward every 4 encoder steps;
3. click button seems to press 2 (or anyway several times) any real click;

I hope this can help you because I choose select a menu library for my project and I'd like choose your library but I cannot use it with this problem.

I have used ClickEncoder for my test.

array81, 2018-11-09:
This is the code used for test:
`#include <Arduino.h>

/********************
Arduino generic menu system
Arduino menu on I2C LCD example
http://www.r-site.net/?at=//op%5B%40id=%273090%27%5D

Sep.2014 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com

LCD library:
https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home
http://playground.arduino.cc/Code/LCD3wires

LCD SCL -> D1
LCD SDA -> D2
ENCODER CLK -> D3
ENCODER DT -> D4
ENCODER SW -> D5

*/
extern "C" {
#include "user_interface.h"
}

#ifndef ARDUINO_SAM_DUE

  #include <Wire.h>
  #include <LiquidCrystal_I2C.h>//F. Malpartida LCD's driver
  #include <menu.h>//menu macros and objects
  #include <menuIO/lcdOut.h>//malpartidas lcd menu output
  #include <menuIO/serialIn.h>//Serial input

  #include <ClickEncoder.h>
  #include <menuIO/clickEncoderIn.h>
  //#include <menuIO/encoderIn.h>//quadrature encoder driver and fake stream
  #include <menuIO/keyIn.h>//keyboard driver and fake stream (for the encoder button)
  #include <menuIO/chainStream.h>// concatenate multiple input streams (this allows adding a button to the encoder)

  using namespace Menu;

  //LiquidCrystal_I2C lcd(0x27);//, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address
  LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address and pinout

  // Encoder /////////////////////////////////////
  #define encA D3
  #define encB D4
  //this encoder has a button here
  #define encBtn D5

  //encoderIn<encA,encB> encoder;//simple quad encoder driver
  //#define ENC_SENSIVITY 4
  //encoderInStream<encA,encB> encStream(encoder,ENC_SENSIVITY);// simple quad encoder fake Stream

  #define encSteps 4
  ClickEncoder clickEncoder = ClickEncoder(encA,encB,encBtn,encSteps);
  ClickEncoderStream encStream(clickEncoder,1);

  //esp8266 timer
  os_timer_t myTimer;
  void timerCallback(void *) {clickEncoder.service();}

  //a keyboard with only one key as the encoder button
  keyMap encBtn_map[]={{-encBtn,defaultNavCodes[enterCmd].ch}};//negative pin numbers use internal pull-up, this is on when low
  keyIn<1> encButton(encBtn_map);//1 is the number of keys

  serialIn serial(Serial);

  //input from the encoder + encoder button + serial
  menuIn* inputsList[]={&encStream,&encButton,&serial};
[... 151 lines cut ...]
    Serial.println("Arduino Menu Library");Serial.flush();
    //encoder.begin();
    lcd.begin(20,4);
    nav.idleTask=idle;//point a function to be used when menu is suspended
    mainMenu[1].enabled=disabledStatus;
    nav.showTitle=false;
    lcd.setCursor(0, 0);
    lcd.print("Menu 4.x LCD");
    lcd.setCursor(0, 1);
    lcd.print("r-site.net");

    delay(2000);

    //encoder timer!
    os_timer_setfn(&myTimer, timerCallback, NULL);//set the callback funtion
    os_timer_arm(&myTimer, 1, true);//setup the timer tick
  }

  void loop() {
    nav.poll();
    digitalWrite(LEDPIN, ledCtrl);
    //delay(100);//simulate a delay as if other tasks are running
  }

#endif`

lorol, 2018-12-26:
Hi @array81 
I noticed your encoder signals are like this:

ENCODER CLK -> D3   ???
ENCODER DT -> D4     ???
ENCODER SW -> D5

So what kind of encoder device you are using? 
The library (I believe) expects simply de-phased square pulses at both encoder inputs A and B:
Clockwise turns
А:____П_П_П__
B: ____П_П_П__
Counter Clockwise turns
А: ____П_П_П__
B:____П_П_П___

What your encoder' CLK and DT signals look like?

neu-rah, 2018-12-27:
hi!
I'm using an quad-encoder with 1 button, so 2 pins are for encoder (out of phase as mentioned) and another for the button, there together for convenience but the driver for encoder is independent from the button, if using the internal encoder driver. However there is an alternative driver, click-encoder that can also be used.

**internal encoder driver**, uses PCINT library to handle pin-change interrupts, can use all atmega pins... but might not go well with all mcus.

**click-encoder** driver uses a timer to handle the encoder

array81, 2018-12-27:
Hi,
I'm using a KY-040, please see here for details: http://henrysbench.capnfatz.com/henrys-bench/arduino-sensors-and-input/keyes-ky-040-arduino-rotary-encoder-user-manual/

These are the connections on NodeMCU:
ENCODER CLK -> D3
ENCODER DT -> D4
ENCODER SW -> D5

These are the connections on Arduino Nano:
ENCODER CLK -> D2
ENCODER DT -> D3
ENCODER SW -> D4

As said with Arduino it works correctly but with NodeMCU there are many problems:
- CW rotation works;
- CCW rotation NOT works;
- CLICK works only sometimes;

lorol, 2018-12-27:
Hi @neu-rah 
First, thank you for the great piece of work - the ArduinoMenu!

Yes I understood the driver difference. My only point was to check the hardware. 
 ESP8266 has few IO pins, some are used for programming, some have pull-ups or not, depending of the board. If both encoder signals are not well formed (example missing pull-up) - it may cause inconsistency.

I am not very clear of the situation - is your (@neu-rah ) setup working as expected and only @array81 experiences problems or the issue was found by both of you ?

lorol, 2018-12-27:
@array81 
Hi, I think NodeMCU D5 (GPIO14) has no pull-up resistor on board.

[image]

array81, 2018-12-27:
> I am not very clear of the situation - is your (@neu-rah ) setup working as expected and only @array81 experiences problems or the issue was found by both of you ?

I think @neu-rah don't have a NodeMCU so he did not test.

> Hi, I think NodeMCU D5 (GPIO14) has no pull-up resistor on board.

In this moment I cannot show the image (I'm not at home). But if you're right I should change pin then test it.

neu-rah, 2018-12-27:
I'm using clickencoder (timer driven) for mcu and internal from atmegas, and yes i didn't gave a try for internal driver on mcu (yet) :smile: 
thanks for the interest, there is also the chat room about ArduinoMenu https://gitter.im/ArduinoMenu/Lobby

lorol, 2018-12-28:
Hi,
I managed to run the example by timer-driven ClickEncoder on NodeMCU.
You need a good rotary encoder and make sure pull-up resistors present on your board or externally on encA, encB and button lines, 
My encoder was old and inconsistent and finally broke so I had to build and encoder from mouse wheel to finish my tests :) 
I observed a button reaction delay when using ClickEncoder library for it.
Better to use the separate "keyboard" alternative with one key for button. 
(clickencoder only for +/- UP/DOWN functionality)

So ... here are the hints for ESP8266 timer:
```
// add this at beginning 
extern "C" {
#include "user_interface.h"
}
...
// create timer and callback 
os_timer_t myTimer;
void timerCallback(void *)
{
  clickEncoder.service();
}
...

void setup() {
...
  //initialize & run the Timer
  os_timer_setfn(&myTimer, timerCallback, NULL);//set the callback function for the encoder
  os_timer_arm(&myTimer, 2, true);//setup the timer tick to call it every 2 ms
  //clickEncoder.setAccelerationEnabled(true);
```

array81, 2018-12-29:
> So ... here are the hints for ESP8266 timer:
I don't understand, with this hints and pull-up resistors it should work on NodeMCU?

I'll have a few tests as soon as possible, these days I do not have much time unfortunately.

lorol, 2018-12-29:
@array81 Hi Martin,
Yes, with these hints, pull-ups and good encoder NodeMCU runs smooth with code based on U8G2.ino Here are the pins of my test, in (L/H) are the levels required by NodeMCU to boot properly:
I have *4 x free DIO pins 3V logic remaining to use to control my external devices + one analog A0 input.
RX and TX are reserved for serial and programming, but it can be used too (with caution).

*D0   (L)                has a pull-down on NodeMCU PCB
D1   SCL   of SSD1306 128x64 I2C OLED
D2   SDA  of SSD1306 128x64 I2C OLED
*D3   (H)                has a pull-up on NodeMCU PCB
D4   (H) enc SW    has a pull-up on NodeMCU PCB
*D5
D6  encA             I added external 10k pull-up
D7  encB             I added external 10k pull-up
*D8                      has pull-down on NodeMCU PCB
D9 (RX)
D10 (TX)
*A0                     analog input

array81, 2018-12-30:
I have try to use you suggestion on IC2 (see post 9 Nov) so I have change pin and add pull-up (timer code is just ok) but the result is the same counterclockwise works always but clockwise and click button no.
I am sincerely thinking of abandoning NodeMCU, my problem is Arduino Nano have very small memory and using ArduinoMenu little space remains.

neu-rah, 2018-12-30:
you can use code driven menu, by handling the encoder with the driver of your choice and issuing menu commands to the navigation root.
I've been using wemos D1 mini with clickencoder quite well...
and yes atmegas are short on memory when using the menu, even reducing numeric field to be all of the same time (ie: int)

note: every usage of field with a different type instantiates the fields code for that type (or part of them).

Ulli2k, 2019-07-14:
I have the same issue on the ESP32 and the rotary encoder (https://www.az-delivery.de/products/drehimpulsgeber-modul?ls=de) which has pullup resistors preinstalled.

neu-rah, 2019-07-14:
@Ulli2k what driver are you using? the internal driver (using interrupts) or the clickEncoder driver (using timer)

also make sure that internal MCU pull-ups are not messing  the signals and that pins are ok, swaping an encoder pin with the button pins will have undesirable results

I've used clickencoder with esp8266, let me know if you need help setting up the timer.

the internal driver does not do software debounce, so it should be used with hw debounce (a couple of small capacitors between the encoder pins and ground)

orlo11, 2019-10-03:
Hello,
**@neu-rah**  how did you manage to use ClickEncoder on ESP8622? Fails to compile for me because I lack the /avr/* includes in this lib. I commented them and could compile and upload my sketch (adapted TFT_eSPI example), but the encoder seams not to work. 
I use the ESP8622-timer like array81 suggested above and a KY-040 encoder. I have to use different pins (D6/D1 for the encoder, D4 for the button) because of the TFT display and D2 and D0 will be consumed by other hardware in my project.

orlo11, 2019-10-03:
I forgot to say -- I use a WEMOS D1 mini, so I name the pins in my sketch the way above -- they are the GPIOs 12/5 for the encoder and 2 for the button.

neu-rah, 2019-10-03:
@orlo11 just comment out or delete those lines, the functionality is already included on esp8266

dboon1978, 2019-11-11:
Is it possible to use a different rotary encoder library?  I've no problems using the ESPRotary library with multiple pin setups on the ESP32, and it would make it a lot easier if I could just assign those controls to interact with this menu library.

chamroeunbm, 2021-03-01:
Same issue with my **ESP32 WROOM**(**The button work but both CW and CCW are jumping around or sometimes, simply no action**) and on Arduino MEGA it works. It try this library and the encoder work fine: https://github.com/igorantolic/ai-esp32-rotary-encoder on my ESP32. So I guess, it's software issue.

Any suggestion or how to integrate ai-esp32 library into this library?

neu-rah, 2021-03-01:
yes, you can check you encoder state and issue menu commands as required, see the codectrl example here https://github.com/neu-rah/ArduinoMenu/blob/master/examples/codeCtrl/codeCtrl/codeCtrl.ino#L168

chamroeunbm, 2021-03-02:
ok, this codeCtrl sample works, but it is only one direction CW and its library is different from the LCD i2c sample. so how can i merge it with the crystal lcd i2c sample? thanks

neu-rah, 2021-03-02:
code control allows any input, you must handle your input and do your encoder reading and decide what menu command to call

here is an enumeration of commands: https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuBase.h#L89

chamroeunbm, 2021-03-03:
Sorry, I am new to this thing.

Since codeCtrl have only 1 directional button and a click button, i add another NAV_BTN2 which point to another PIN. but it always call the NAV_BTN for both up and down rotation.

Please note that I use rotary encoder.

(!digitalRead(SEL_BTN)) {
    delay(SOFT_DEBOUNCE_MS);
    while(!digitalRead(SEL_BTN));
    nav.doNav(enterCmd);
    delay(SOFT_DEBOUNCE_MS);
  }
  if (!digitalRead(NAV_BTN)) {
    delay(SOFT_DEBOUNCE_MS);
    while(!digitalRead(NAV_BTN));//wait for button release
    nav.doNav(upCmd);
    delay(SOFT_DEBOUNCE_MS);
    Serial.println("UP");
    
  }
/**I add this this function and bind it to another PIN of my rotary encoder**/
  if (!digitalRead(NAV_BTN2)) {
    delay(SOFT_DEBOUNCE_MS);
    while(!digitalRead(NAV_BTN2));//wait for button release
    nav.doNav(downCmd);
    delay(SOFT_DEBOUNCE_MS);
    Serial.println("DOWN");
  }

neu-rah, 2021-03-03:
here is the wiki section about it.
https://github.com/neu-rah/ArduinoMenu/wiki/Navigation#enum-navcmds

when using commands the menu system wont read you encoder and you should NOT configure the encoder as an input.
you must read the encoder using its on drive and issue menu commands to the nav root according to you encoder readings... so you will be 100% responsible for the relation between the encoder and the menu.

the code section you pasted above is just a demo using software debounce on some pins. Because the menu key driver has no debounce we are reading the pins and doing the debounce and eventually send menu commands to do some menu action.

chamroeunbm, 2021-03-03:
the thing is i cannot find any working rotary encoder driver that work with my esp32 from your sample yet.

this git work https://github.com/igorantolic/ai-esp32-rotary-encoder but i have no idea how to integrate with your menu

neu-rah, 2021-03-04:
if you have any encoder driver that work **without** the menu, then you can use a menu (without input) and read your working driver then issue menu commands to your intended encoder action.

chamroeunbm, 2021-03-05:
you mean use the working driver and call your menu code or we can say merge the other encoder library with your menu?