neu-rah/ArduinoMenu #211: ESP8266WiFi.h: No such file or directory

rahulhk04, 2018-10-07:
ESP8266WiFi.h: No such file or directory
i have added library files 
but still i am getting this problem please resolve this problem 

Code For Arduino 
#include "ACS712.h"
char watt[5];
ACS712 sensor(ACS712_30A, A0);
unsigned long last_time =0;
unsigned long current_time =0;
float Wh =0 ;  
void setup() {
  Serial.begin(115200);
  sensor.calibrate();
}

void loop() {
  float V = 230;
  float I = sensor.getCurrentAC();
// Serial.println(I);
  float P = V * I;
  last_time = current_time;
  current_time = millis();    
  Wh = Wh+  P *(( current_time -last_time) /3600000.0) ; 
   dtostrf(Wh, 4, 2, watt);        
Serial.write(watt);
  delay(10000);
}

 

Code For NodeMCU
#include <ESP8266WiFi.h>
#include "Adafruit_MQTT_Client.h"
#define WLAN_SSID       "a*************"
#define WLAN_PASS       "*******************"
char watt[5];
#define AIO_SERVER      "io.adafruit.com"
#define AIO_SERVERPORT  1883                   
#define AIO_USERNAME    "rjrishabh"
#define AIO_KEY         "***********************"

WiFiClient client;
int bill_amount = 0;   
unsigned int energyTariff = 8.0; 
Adafruit_MQTT_Client mqtt(&client, AIO_SERVER, AIO_SERVERPORT, AIO_USERNAME, AIO_KEY);
Adafruit_MQTT_Publish Power = Adafruit_MQTT_Publish(&mqtt, AIO_USERNAME "/feeds/Power");
Adafruit_MQTT_Publish bill = Adafruit_MQTT_Publish(&mqtt, AIO_USERNAME "/feeds/bill");
void MQTT_connect();

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.println(F("Adafruit MQTT demo"));

  // Connect to WiFi access point.
  Serial.println(); Serial.println();
  Serial.print("Connecting to ");
  Serial.println(WLAN_SSID);

  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  Serial.println("WiFi connected");
  Serial.println("IP address: "); Serial.println(WiFi.localIP());
}

void loop() {
  // Ensure the connection to the MQTT server is alive (this will make the first
  // connection and automatically reconnect when disconnected).  See the MQTT_connect
  // function definition further below.
  MQTT_connect();

 int i=0;
 float watt1;
 if(Serial.available() > 0 ){
   delay(100); //allows all serial sent to be received together
    while(Serial.available() && i<5) {
     watt[i++] = Serial.read();
    }
    watt[i++]='\0';
  }

 watt1 = atof(watt);
  bill_amount = watt1 * (energyTariff/1000);      // 1unit = 1kwH
  Serial.print(F("\nSending Power val "));
  Serial.println(watt1);
  Serial.print("...");

  if (! Power.publish(watt1)) {
    Serial.println(F("Failed"));
  } else {
    Serial.println(F("OK!"));
  }

   if (! bill.publish(bill_amount)) {
    Serial.println(F("Failed"));
  } else {
    Serial.println(F("OK!"));
  }

if (bill_amount==4){
for (int i =0; i<=2; i++)
{
  bill.publish(bill_amount);
delay(5000);
}
bill_amount =6;
}
  
delay(5000);
}

// Function to connect and reconnect as necessary to the MQTT server.
// Should be called in the loop function and it will take care if connecting.
void MQTT_connect() {
  int8_t ret;

  // Stop if already connected.
  if (mqtt.connected()) {
    return;
  }

  Serial.print("Connecting to MQTT... ");

  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
       Serial.println(mqtt.connectErrorString(ret));
       Serial.println("Retrying MQTT connection in 5 seconds...");
       mqtt.disconnect();
       delay(5000);  // wait 5 seconds
       retries--;
       if (retries == 0) {
         // basically die and wait for WDT to reset me
         while (1);
       }
  }
  Serial.println("MQTT Connected!");
}

rahulhk04, 2018-10-07:
iam using this for arduino nano . while compiling this program i am getting this issue  ESP8266WiFi.h: No such file or directory . 
link  https://github.com/adafruit/Adafruit_MQTT_Library

neu-rah, 2018-10-07:
I'm using this core + libs:
https://github.com/esp8266/Arduino

the esp8266wifi.h must be on this specific lib https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi

rahulhk04, 2018-10-07:
am not geting the download link 
pleas shear me the download link of the library

neu-rah, 2018-10-07:
ah, arduino **nano**, sorry i skiped that part, you wont have that include while compiling for arduino nano, the esp8266 examples are meant to run on bare esp8266, but you can use any other example for nano

closed 2019-01-21 by neu-rah