neu-rah/ArduinoMenu #212: Error ven compiling the u8g2 example

AndersV209, 2018-10-29:
Hello 
i get this error when i try to compile the example sketch for the u8g2 libary 
i am using and esp8266 board 
`Archiving built core (caching) in: C:\Users\Ander\AppData\Local\Temp\arduino_cache_364406\core\core_esp8266_esp8266_generic_CpuFrequency_80,VTable_flash,ResetMethod_nodemcu,CrystalFreq_26,FlashFreq_40,FlashMode_qio,FlashSize_4M1M,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_39f62cb8e066730206522120b3683e53.a
C:\Users\Ander\AppData\Local\Temp\arduino_build_558914/arduino.ar(time.c.o): In function `time':

C:\Users\Ander\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/time.c:81: multiple definition of `time'

sketch\U8G2.ino.cpp.o:C:\Users\Ander\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Print.h:93: first defined here

c:/users/ander/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: Warning: size of symbol `time' changed from 16 in sketch\U8G2.ino.cpp.o to 29 in C:\Users\Ander\AppData\Local\Temp\arduino_build_558914/arduino.ar(time.c.o)

c:/users/ander/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: Warning: type of symbol `time' changed from 1 to 2 in C:\Users\Ander\AppData\Local\Temp\arduino_build_558914/arduino.ar(time.c.o)

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8266 Module.`

why does it not work

neu-rah, 2018-10-30:
there seem to be a conflict, two libraries providing the `time` function

C:\Users\Ander\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/time.c:81: multiple definition of `time', i wil try to compile the example for esp8266 here, seems you are using arduino ide, what version?

AndersV209, 2018-11-03:
i am using arduino ide version 1.8.5

AndersV209, 2018-11-03:
i changed the time to time2 and it could compile

TamojitSaha, 2018-11-07:
Please close the issue, if you are satisfied.

closed 2019-01-21 by neu-rah