neu-rah/ArduinoMenu #213: Macros for SSD1306ASCII_OUT missing

hopfermann, 2018-11-01:
Hello,

if I am not mistaken the following macros are missing from macros.h. I am not sure if I got everything right, but it compiles OK for me.

#define VAR_SSD1306ASCII_OUT(id,md,n,oled,fontW,fontH,...)\
Menu::idx_t id##Tops##n[md];\
PANELS(id##Panels##n,__VA_ARGS__);\
Menu::SSD1306AsciiOut id##n(&oled,id##Tops##n,id##Panels##n,fontW,fontH);

#define REF_SSD1306ASCII_OUT(id,md,n,...) &id##n,

Best regards
Hartmut

neu-rah, 2018-11-02:
Hi!
Yes, some displays are missing the build macro, a pull request would be welcome :D

hopfermann, 2018-11-02:
This would be my first pull request, so I am not quite sure (even after reading the documentation) how to go about it. Would I need to create my own branch or could I use an existing one? Sorry, but I am a complete GitHub newbie (apart from grabbing source code, that is) ...

neu-rah, 2018-11-03:
*not being and expert on it*

I think you need to fork the repo, do the changes on your clone and when happy do a pull request

TamojitSaha, 2018-11-07:
> 
> 
> This would be my first pull request, so I am not quite sure (even after reading the documentation) how to go about it. Would I need to create my own branch or could I use an existing one? Sorry, but I am a complete GitHub newbie (apart from grabbing source code, that is) ...

Just fork the repo into your repository, make the necessary changes, then from your forked repository click on the pull request. Thats it. Its simple

rogerjames99, 2023-04-24:
For anyone seeing this. To get this working on a more recent compiler you need to update VA_ARGS to __VA_ARGS__.

Also the continuation line markers are missing. The code should look like this.

#define VAR_SSD1306ASCII_OUT(id,md,n,oled,fontW,fontH,...)\
Menu::idx_t id##Tops##n[md];\
PANELS(id##Panels##n,__VA_ARGS__);\
Menu::SSD1306AsciiOut id##n(&oled,id##Tops##n,id##Panels##n,fontW,fontH);

#define REF_SSD1306ASCII_OUT(id,md,n,...) &id##n,

I will raise a pull request.