neu-rah/ArduinoMenu #216: btnNav not responding to buton inputs

mbc99, 2018-11-06:
If've set the circuit in order to recreate the btnNav example. So I've connected pins 6,7,8 to switch using pull down resistors. Before trying the code I've made a code to see that the arduino is really seeing the switch get touched: 
```
void setup() {
  Serial.begin(9600);
  pinMode(6, INPUT);
  pinMode(7, INPUT);
  pinMode(8, INPUT);
}

void loop() {
  Serial.print(digitalRead(6));
  Serial.print(digitalRead(7));
  Serial.println(digitalRead(8));
}

```
But when I upload the btnNav sketch up and down buttons don't respond. However, the select button does respond when I click the switch (At the end I removed the switch and use a simple wire connected (or not) to 5V to simulate the switch).

I've also tried changing the input pins. As the select button pin seems to work I've assigned that pin to the down button, but it still won't work (only the select button).

It's curious because when I click the select button the arduino TX led blinks, but when I click the other two switch nothing happens

TamojitSaha, 2018-11-07:
Hi,
You dont require input pull-down as because in the library the buttons internally pulled-up.

In the following lines...
https://github.com/neu-rah/ArduinoMenu/blob/06eaf3fabaff68d17f87035627f83957f1340ee6/examples/btnNav/btnNav/btnNav.ino#L111-L115

`-`indicates internal pullup.
You can also refer to [this](https://github.com/neu-rah/ArduinoMenu/tree/master/examples/SSD1306Ascii/Button_Navigation) sketch for more details.

mbc99, 2018-11-07:
Cool, it was obviously that.
However, it would be nice that this was indicated in the pin number declaration (#define....) (I don't know if it was stated somewhere else and I haven't read it)

TamojitSaha, 2018-11-09:
@mbc99 
Nice to hear that your issue is solved. Cheers!! 🥂 😉 
As  I said earlier, you can refer to [this](https://github.com/neu-rah/ArduinoMenu/tree/master/examples/SSD1306Ascii/Button_Navigation) button navigation example for thorough understanding.

https://github.com/neu-rah/ArduinoMenu/blob/06eaf3fabaff68d17f87035627f83957f1340ee6/examples/SSD1306Ascii/Button_Navigation/config.h#L47

TamojitSaha, 2018-11-30:
@mbc99
Is you issue solved?
If yes, please close this issue.

closed 2018-11-30 by mbc99