neu-rah/ArduinoMenu #219: Menu hanging after second use

slave2anubis, 2018-11-14:
Hello, i am haveing a problem, that i think its maybe a bug.
So: i am doing a project useing a Arduino Nano, a Oled 128x64 screen and a encoder with select function.
I wired everything and got the project to work, but there is one big problem.
After i power up the board i set a "flag" to false so that the menu is not shown, after i setup a event handled on button select to set the flag to true and display the menu. 
The first time, everything works as intended, on boot i get my screen and when i press the select the menu screen is shown and everything works as expected. After i exit the menu and go back to my screen, enverything normal i see my screen it updates as normal, but when i try to enter the menu again using the select button, the screen hangs and nothing happens, i click again, and i get back to my screen, not the menu, after i click again, and sometimes it hangs but moveing the +/- direction the screen "un-hangs" and i see the menu.

I know this problem is a bit weird to understand, but maybe someone can help!
`#include <Arduino.h>
#include <Wire.h>
#include <menu.h>//menu macros and objects
#include <Adafruit_MCP4725.h>
#include "MCP3221.h"
#include <ClickEncoder.h>
#include <TimerOne.h>
#include <menuIO/clickEncoderIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>
#include <menuIO/U8x8Out.h>
#include <U8x8lib.h>

#define LEDPIN LED_BUILTIN

#define DEBUG

#ifndef DEBUG
Print& operator<<(Print&o, Menu::prompt&p) {
  print_P(o,p.getText());
  return o;
}
#endif

const byte DEV_ADDR = 0x4A;

Adafruit_MCP4725 dac; // constructor
MCP3221 mcp3221(DEV_ADDR);

U8X8_SH1106_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);

// These constants won't change. They're used to give names to the pins used: 
const int analogInPin = A6;  // Analog input pin that the potentiometer is attached to
int analogamount = 0; // stores incoming value

float voltage = 0; // used to store voltage value

int sensorValue = 0;        // value read from the pot
int outputValue = 0;        // value output to the PWM (analog out)

const int numReadings = 10;

int readings[numReadings];
int readIndex = 0;              // the index of the current reading
int total = 0;                  // the running total
int average = 0;                // the average
int next = 0;
int uInRaw = 0;
unsigned int userCurrSetRaw = 0;
unsigned int currentReadRaw = 0;
unsigned int newUserCurrSetRaw = 0;
double currentReadmA = 0;
double currentReadA = 0;
double userCurrSet, resistor;
double capacitymAh = 0;
double discharge = 0;
int timer = 0;
int oneSecond = 0;
int test=55;

boolean runTestFlag = false;
boolean menuShowFlag = false;

float uIn = 0;

//encoder
#define encA 4
#define encB 2
//this encoder has a button here
#define encBtn 3

ClickEncoder clickEncoder(encA,encB,encBtn,2);
ClickEncoderStream encStream(clickEncoder,1);
MENU_INPUTS(in,&encStream);

void timerIsr() {
  clickEncoder.service();
  timer++;
  oneSecond++;
}

//Menu
MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,FIELD(test,"Test","%",0,100,10,1,doNothing,noEvent,wrapStyle)
  ,EXIT("<Back")
);

#define MAX_DEPTH 1

MENU_OUTPUTS(out,MAX_DEPTH
  ,U8X8_OUT(u8x8,{0,0,16,8})
  ,SERIAL_OUT(Serial)
);

NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);//the navigation root object

result idle(menuOut& o,idleEvent e) {
  menuShowFlag = false;
  return proceed;
}

void setup() {
  pinMode(encBtn,INPUT_PULLUP);
  Serial.begin(115200);
  u8x8.begin();
  u8x8.setFont(u8x8_font_chroma48medium8_r);

  nav.idleTask=idle;//point a function to be used when menu is suspended

  mcp3221.reset();
  mcp3221.setVref(4096);
  mcp3221.setSmoothing(ROLLING_AVG);
  mcp3221.setNumSamples(15);
  
  analogReference(EXTERNAL);

  dac.begin(0x60);
  dac.setVoltage(userCurrSet, false);

  resistor = 0.25;

  Timer1.initialize(1000);
  Timer1.attachInterrupt(timerIsr); 
  
  // initialize all the readings to 0:
  for (int thisReading = 0; thisReading < numReadings; thisReading++) {
    readings[thisReading] = 0;
  }
}

void loop() {  

  //activate the menu when the select button is pressed
  if(!menuShowFlag){
    ClickEncoder::Button b = clickEncoder.getButton();
    if (b != ClickEncoder::Open) {
      if(b == ClickEncoder::Clicked){
          menuShowFlag = true;
        }
      }
   }

   if(menuShowFlag){
      nav.poll();
   }

 
  if(timer >= 50){
    currentReadRaw = getVoltageAdcExt();
    timer = 0;

  }
  
  if(oneSecond >= 1000){
    oneSecond = 0;
    if(runTestFlag){
      discharge += currentReadmA;
      }
  }

  if(discharge > 0){
    capacitymAh = discharge/3600;    
    }
  

  uInRaw = getReading();
  uIn = uInRaw * 0.0213;

  // Prevent over discharge
  if(runTestFlag){
    if(uIn <= 3.2){
      userCurrSetRaw = 0;
      setCurr(userCurrSetRaw);
      runTestFlag = false;
      }
  }

  userCurrSet = (userCurrSetRaw / resistor) / 1000;
  currentReadmA = currentReadRaw / resistor;
  currentReadA = currentReadmA / 1000;  

  if(!menuShowFlag){
    if(next >= 20){
      lcdOut(userCurrSet, currentReadA, uIn, capacitymAh);    
      next = 0;
      }
    next++; 
    }

   
}
void lcdOut(float currentSet, float currentRead,  float uIn, float capacitymAh){
    u8x8.setCursor(0, 0);
    u8x8.print("CurrSet:");
    u8x8.print(currentSet);
    u8x8.println("A");  
    u8x8.print("CurrRead:");
    u8x8.print(currentRead);
    u8x8.println("A");
    u8x8.print("U:");
    u8x8.print(uIn);
    u8x8.println("V");
    u8x8.print("Cap:");
    u8x8.print(capacitymAh);
    u8x8.println("mAh");
    
  }

int getVoltageAdcExt(){
  return mcp3221.getVoltage();
  }

void setCurr(int current){
  dac.setVoltage(current, false);
  }

int getReading(){

   for(byte i = 0; i <= numReadings; i++){
    // subtract the last reading:
    total = total - readings[readIndex];
    // read from the sensor:
    readings[readIndex] = analogRead(analogInPin);
    // add the reading to the total:
    total = total + readings[readIndex];
    // advance to the next position in the array:
    readIndex = readIndex + 1;
  
    // if we're at the end of the array...
    if (readIndex >= numReadings) {
      // ...wrap around to the beginning:
      readIndex = 0;
    }
  
    // calculate the average:
    average = total / numReadings;
    // send it to the computer as ASCII digits

   }
   return average;
  }`

vladkozlov69, 2018-11-14:
Hello,

If you want to start your application without menu, just call nav.idleOn()
in the end of setup() method; the flag is not required. It would be better
to give full control on encoder to menu system instead of calling encoder
API.

Regards,
Vlad

neu-rah, 2018-11-16:
hi! :D
not sure if i'm getting it right (and not read the code yet), but it looks like you use your own code to display info on the screen... you should clean the screen before reentering the menu, the menu will only draw if you move the cursor because its not aware of your screen usage, there is a method to force the menu redraw on reenter... but you can also use idling functions to display your info, in that case the menu will auto resume on enter/esc and will clear screen and redraw by it self, see idling on wiki https://github.com/neu-rah/ArduinoMenu/wiki/Idling

d21d3q, 2019-01-19:
I am also observing menu crash when menu goes to idle from submenu and when resetting nav from iddle task (`nav.reset()`) or when resetting nav from loop (using flag indicating that menu should reset. 
Desired result is, that when going back from idle, menu should start from root.

neu-rah, 2019-01-19:
@d21d3q what do you have in common with this?:
```c++
#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>
#include <menuIO/U8x8Out.h>
```

d21d3q, 2019-01-19:
```C
// those
#include <menuIO/encoderIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>

// aditionally
#include <menuIO/liquidCrystalOut.h>
// those are included, but currently not used
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>
```

I am using nucleo fo91rc, half of memory is used.
I didn't manage to enable debugging (yet) on this platform, so that I'll try to reproduce it on avr and provide some minimal code.

neu-rah, 2019-01-19:
@slave2anubis you are using a blocking function to execute your code on the batt.->start option, that means the menu does not expect you to change the screen, but you can. So, now you just have to signal the menu system that it need to redraw the menu (you might also need to clean up the screen before returning)

please try adding this to your handler
`batteryCap.dirty=true`

d21d3q, 2019-01-19:
Idle function is receiving `menuOut` object, screen should be cleaned with it via `o.clear()`?

neu-rah, 2019-01-19:
i mean on this part (battery status), it is blocking the menu system and using the screen at will, perfectly ok, but the menu is just frozen and away from mcu... or so it look at a first glance to the code :D

```c++
    while(true){
       battAnalyzeFinished(capacitymAh);
       ClickEncoder::Button b = clickEncoder.getButton();
       if (b != ClickEncoder::Open) {
         if(b == ClickEncoder::Clicked){
              setCurrentmA(0);
              return proceed;
            }
          }
      }
   
```

d21d3q, 2019-01-20:
You can reproduce error with [serial example](https://github.com/neu-rah/ArduinoMenu/tree/master/examples/Serial) by enabling idle function and calling `nav.reset()` on `idleEnd` event.

```C
result idle(menuOut &o, idleEvent e) {
  // o.clear();
  switch(e) {
    case idleStart:o.println("suspending menu!");break;
    case idling:o.println("suspended...");break;
    case idleEnd:o.println("resuming menu.");
      nav.reset();
      break;
  }
  return proceed;
}
```
Now enter `Choose` menu and wait for idle. after entering menu again you will notice strange behavior. Stm32 will freeze, avr will continue, sometimes printing garbage.

```
Main menu
[1] Op1
[2] Op2
[3] Test 55.00%
[4] Sub-Menu
[5] Led: Off
[6] LED On
[7] LED Off
[8] Select Zero
[9]>Choose Last

Choose
[1] First
[2] Second
[3] Third
[4]>Last       // wait for idle 
suspending menu!
suspended...
resuming menu.    // input: *

Main menu
[1]>Op1
[2] Op2
[3] Test 55.00%
[4] Sub-Menu
[5] Led: Off
[6] LED On
[7] LED Off
[8] Select Zero
[9] Chooseevent: 32

// input: +
Main menu  
[1] Op1
[2] Op2
[3] Test 55.00%
[4] Sub-Menu
[5]>Led: Off             // notice jump from 1 to 5
[6] LED On
[7] LED Off
[8] Select Zero
[9] Choose16 action1 executed, proceed menu
```
With select menu: 
```
Main menu
[1] Op1
[2] Op2
[3] Test 55.00%
[4] Sub-Menu
[5] Led: Off
[6] LED On
[7] LED Off
[8]>Select:Zero
[9] Choose Last
suspending menu!
suspended...
resuming menu.

Main menu
[1]>Op1
[2] Op2
[3] Test 55.00%
[4] Sub-Menu
[5] Led: Off
[6] LED On
[7] LED Off
[8] Select Zero
[9] Choose Last32 action1 executed, proceed menu

Main menu
[1] Op1
[2]>Op2
[3] Test 55.00%
[4] Sub-Menu
[5] Led: Off
[6] LED On
[7] LED Off
[8] Select One
[9] Choose Last

Main menu
[1] Op1
[2] Op2
[3]>Test 55.00%
[4] Sub-Menu
[5] Led: Off
[6] LED On
[7] LED Off
[8] Select Two
[9] Choose Lastevent: 16

Main menu
[1] Op1
[2] Op2
[3] Test 55.00%
[4]>Sub-Menu
[5] Led: Off
[6] LED On
[7] LED Off
[8] Select w⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ϓߓ
[9] Choose Last16 action1 executed, proceed menu
```
Looks like wild pointers.

closed 2019-01-21 by neu-rah