neu-rah/ArduinoMenu #221: Using AdafruitGfxOut how to implement (GxEPD2)

Jacotheron, 2018-11-29:
On my project, I am using an eInk display, controlled with the AdafruitGFX library (more specifically the GxEPD2 library, which implements the AdafruitGFX library).

I have defined everything and have the display working, I have also defined my menu and inputs correctly. The part I am struggling with is defining the display as a Menu Output. It seems to me that because it is the GxEPD2 object, the AdafruitGfxOut can't handle it.

How do I define a new Output for this?

I know that eInk displays are slow to refresh etc, but that is simply what my project asks for. The menu will mostly be used to setup and modify parameters that can't be done remotely (BLE interface) - so not regularly enough to justify an extra display (also there is no pins left, for such a display, and wires are too long for I2C display).

neu-rah, 2019-01-21:
@Jacotheron can you paste a link to the library you use?
I'm finding multiple sources

Jacotheron, 2019-01-21:
In Arduino IDE Library Manager, its name is "ArduinoMenu library" by "Rui Azevedo, ruihfazevedo" with version 4.15.7.

I believe this is enough info to find it.

Jacotheron, 2019-01-21:
Also in the Library Manager, it is named "GxEPD2" by Jean-Mark Zingg version 1.1.1

Actually in my project, I could not get it working, so I have since added a simple I2C oled for the menu (since I2C does not need extra pins).