neu-rah/ArduinoMenu #222: button issue on oled example

mbc99, 2018-11-30:
Hi, I just tried the oled_button_nav example. It all works fine except one thing: Navigating through the menu is no issue. But when I try to select, say for example the brightness, the buttons are inverted. This means that if for navigating through the menu one button makes the menu go up when selecting the brigthness that same button makes de brightness decrease in value.
Moreover, if I arrange the input pins for the switch like the .h file: 
#define BTN_UP      9                 // Up Button
#define BTN_DOWN    8                 // Down Button

the switch on pin 8 makes the menu go down, but the brightness goes up.

Another thing that happens is that when I try to click on the Led: on/off menu it jumps to the supending-suspended-resuming screens. This only happens the second time I click in that option, and once that happens nothing works. Because once the menu has resumed if I click the up/down button the menu jumps to the supending-suspended-resuming screens.

TamojitSaha, 2018-11-30:
Please post your code if it dientot violates any rights... :wink:

mbc99, 2018-11-30:
`/**************************************************************************
  Sketch: MENU NAVIGATION WITH JUST 4 BUTTONS

  This Sketch displays Menu without any Serial Communication and the
  navigation is performed by 4 buttons attached to D3, D8, D4 and D1.
  Also, attach led on D11 to control brightness from menu.
  Default brightness is 15% (check line 78)

   NOTE: By default, navigation buttons use INTERNAL_PULLUP feature.
         This can be changed by commenting the line 45 in "config.h" file

   Uses SSD1306Ascii Library(https://github.com/greiman/SSD1306Ascii)
   by Bill Grieman

   Created by Tamojit Saha
    Github: https://github.com/TamojitSaha
    Website: https://www.tamojitsaha.info/
***************************************************************************/
#include <Arduino.h>

// #include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#include <menu.h>
#include <menuIO/SSD1306AsciiOut.h>
#include <menuIO/keyIn.h>
//#include <menuIO/chainStream.h>
using namespace Menu;

SSD1306AsciiWire oled;

#include "config.h"

int ledCtrl = HIGH;  //Default LED State of LED at D11 is LOW

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e, navNode& nav, prompt& item) {

  return proceed;
}

result action1(eventMask e) {

  return proceed;
}

result action2(eventMask e, navNode& nav, prompt &item) {
  //Serial.print(e);
  //Serial.println(" action2 executed, quiting menu");
  return quit;
}

result ledOn() {
  ledCtrl = HIGH;
  analogWrite(LED_PIN, 255);
  return proceed;
}
result ledOff() {
  ledCtrl = LOW;
  analogWrite(LED_PIN, 0);
  return proceed;
}

result internalLedOn() {
  digitalWrite(LED_BUILTIN, HIGH);
  return proceed;
}
result internalLedOff() {
[... 131 lines cut ...]
    case idling: o.println("suspended..."); break;
    case idleEnd: o.println("resuming menu."); break;
  }
  return proceed;
}

void setup() {
  joystickBtns.begin();
  pinMode(LED_PIN, OUTPUT);
  pinMode(LED_BUILTIN, OUTPUT);
  Wire.begin();
  oled.begin(&Adafruit128x64, OLED_I2C_ADDRESS); //check config
  oled.setFont(menuFont);
  oled.clear();
  nav.idleTask = idle; //point a function to be used when menu is suspended
}

void loop() {
  nav.poll();
  delay(1);//simulate a delay when other tasks are done
}`

I haven't changed anything from your example. Maybe it's a matter of button orientation. But I would excpect one button to go up on the menu and increase the brightness value.

Thanks :)

TamojitSaha, 2018-11-30:
Please provide me some time to test the code once again in real hardware. I will get back to you within two days. If I forget to post, please leave a comment. :smiley: :wink:

mbc99, 2018-11-30:
No problem :)

Once you are working on it please have a look at how to print a text+variable on .OP option menu (maybe you already know how). At first I thought on using a .FIELD type. But then it is possible for the user to change the value, and this is imposible since I want to display a temperature value from a sensor

TamojitSaha, 2018-12-01:
Sure @mbcc! :Smiley:
I will be working on it.

mbc99, 2018-12-05:
Hi there
Any news? :)

TamojitSaha, 2018-12-05:
Working on it...Hope to resolve it by today

ferchinas, 2018-12-13:
Hi, I have the same problem, comment on it at https://gitter.im/ArduinoMenu/Lobby

I got the following excellent answer:
@ me21
As far as I understand, it is intended. Consider Up button "Plus" instead, it increases current menu index or value.

It seems totally logical, but not intuitive.
I'm also looking if there is any way to change this feedback.

ferchinas, 2018-12-13:
Hi, I found the following issue "https://github.com/neu-rah/ArduinoMenu/issues/140" and the "hard coded" solution works for me.

mbc99, 2019-01-10:
> Hi, I found the following issue "#140" and the "hard coded" solution works for me.

Thanks, I'll take a look at that.
However, I still need to do this:

> No problem :)
>
> Once you are working on it please have a look at how to print a text+variable on .OP option menu (maybe you already know how). At first I thought on using a .FIELD type. But then it [...]

mbc99, 2019-01-22:
> you can,
> 1 - Use a disabled field, user wont be able to change its value but it will still reflect the changing values.
> 2 - Use a custom print prompt that draws the desired values
> 
> we are pretty free on what a custom prompt can draw provided it draws on the correct region of the screen (attributed line and only one line)
> 
> this project uses a custom draw prompt to draw an animated graph inside the menu https://github.com/jarkman/ServoBox
> 
> video of wemos doing it https://youtu.be/JXvVfzJGWxM
> 
> P.S. - disabling a menu item, its a dynamic property you can set it anytime (including setup function)
> 
> `mainMenu[1].disable();` will disable the second option on the mainMenu
> 
> revert it with `.enable()`

Thanks, this issue has been resolved.

> Another thing that happens is that when I try to click on the Led: on/off menu it jumps to the supending-suspended-resuming screens.
With this I have reduced de problem, but is there any way to avoid going to the suspended menu when clicking back on the main menu? I've tried deleting the menu part from the code but it (obviously) continues to try to jump to that menu, now the screen goes blank.

mbc99, 2019-01-22:
> Hi, I found the following issue "#140" and the "hard coded" solution works for me.

I've had a look at that, but I don't know if I'm not looking on the right file. If I go to config.h I don't seem to find that line of command...

neu-rah, 2019-01-22:
hi!
the return value (from a menu event handler) can determine if an exit is to be performed, please make sure your `return proceed;` or `return quit;` from your handler. And yes you can prevent a menu from exit https://github.com/neu-rah/ArduinoMenu/wiki/FAQ#prevent-menu-from-exit

neu-rah, 2019-02-26:
is this solved?

mbc99, 2019-02-26:
I still haven't figured out how to invert the button navigation. The solutions proposed here haven't worked.

mbc99, 2019-04-23:
Well, after giving it a second try I've finally inverted buton navigation. Thanks to @ferchinas :)

I will attach the nav.cpp file with the necessary modifications in case anyone wants to do the same :)

closed 2019-04-23 by mbc99