neu-rah/ArduinoMenu #223: Menu Order

BennehBoy, 2018-12-06:
Hi Rui,

Is there any way to change the order of items in a menu from user input?

For example:

A
B
C

User presses select on item B, then use of up or down inputs followed by select again moves it in the list.

So select, up, select would result in:

B
A
C

Is this something that would be useful as base menu system functionality, or do I need to try and encode this myself using a dynamic menu?

neu-rah, 2018-12-06:
Hi! 
**not implemented but possible**, provided that you are using a dynamic menu (non AVR's default) or forcing an AVR to use dynamic by using cmd line defines to the compiler, not recommended any way as AVR's are already low on resources

This library is already huge for avr's and i'm trying not to stretch it further

however the implementation should be a matter of swapping the pointers on the menu data array

we can add inline swap function accepting 2 indexes, if the compiler is smart  enough to ignore it when not used, this of course only for dynamic menus and ignored on static.

this change wont break compatibility :+1:

BennehBoy, 2018-12-07:
Sounds good