neu-rah/ArduinoMenu #224: "Dynamic" example crashes on Arduino Mega 2560

epaterlini, 2019-01-02:
Hello,
I have problem when compiling the "dynamic" example.
I am using Arduino IDE 1.8.8 and Mega2560.
First, it gives an error "This menu example does not work on flash memory versions (MENU_USEPGM) etc etc"...so far so good, so I add the
`#define MENU_USERAM`
directive at the very beginning of the sketch.
Compilation goes fine, but when I run the program, open the serial monitor (it says "press any key") and press a key, it restarts.
I really need this great piece of code working for my project, which is based on dynamic menus using Mega 2560.
I'm interested also on contributing on code if I can help.
Did you see this problem happening on Mega2560?

Thank you very much for your support, happy new year!
Emiliano

neu-rah, 2019-01-03:
Hi!

that directive MUST be on compiler command line otherwise its valid only on the sketch and result in inconsistency and crash, what IDE are you using?

neu-rah, 2019-01-03:
hi, just for a test you can change it here https://github.com/neu-rah/ArduinoMenu/blob/master/src/macros.h#L9

epaterlini, 2019-01-03:
perfect it works flawlessly!
didn't know of this caveat...
sorry for bothering and happy new year :-)

neu-rah, 2019-01-03:
you are welcome, we also have a chat room here: https://gitter.im/ArduinoMenu/Lobby

that caveat is due to the organization of the include files, maybe we can change it but it would require to pre-process excluding all non compatible parts, so that we could have a single include... or do something completely new :D

closed 2019-01-03 by neu-rah