neu-rah/ArduinoMenu #227: Unclear reference to WebSocketsServer.h, file not found

Joennuh, 2019-01-28:
In release 4.15.11 somewhere jsonFmt.h / jsonFmt.cpp is referenced resulting in an error saying WebSocketsServer.h could not befound. When I do a search with Github on the files of the repository I can't find any link to those 3 files. Also I didn't put an include to any of those 3 files.

The error is:
```
In file included from C:\Users\*****\Documents\Arduino\libraries\ArduinoMenu-master\src\menuIO\jsonFmt.h:9:0,

                 from C:\Users\*****\Documents\Arduino\libraries\ArduinoMenu-master\src\menuIO\jsonFmt.cpp:3:

C:\Users\*****\Documents\Arduino\libraries\ArduinoMenu-master\src\menuIO\esp8266Out.h:9:32: fatal error: WebSocketsServer.h: No such file or directory

   #include <WebSocketsServer.h>

                                ^

compilation terminated.
```

The includes I use in my testproject:
```
#include <Arduino.h>
#include <menu.h>
#include <menuIO/u8g2Out.h>
#include <menuIO/serialIn.h>
#include <menuIO/serialOut.h>
#include <menuIO/altKeyIn.h>
#include <menuIO/chainStream.h>
#include <Wire.h>
```

My testproject runs on a Wemos D1 mini clone.

neu-rah, 2019-01-29:
the webmenu example is specific for esp8266 family, the driver includes those file and requires that those libraries be present/installed. Fell free to ask here, on gitter chat room or by email. I'm glad to help.

gitter char room here: https://gitter.im/ArduinoMenu/Lobby

Joennuh, 2019-01-29:
But I didn't use the webmenu example and that's why I'm wondering why the mentioned files are referenced / included. I can't find any reference / link to it.

I have never heard of Gitter but I signed up on it now.

neu-rah, 2019-02-03:
must be solved, https://github.com/neu-rah/ArduinoMenu/commit/823a32a6204655ecc3501af004d9d27a3c8c1045

closed 2019-02-03 by neu-rah