neu-rah/ArduinoMenu #228: streamFlow.h: No such file or directory

Joennuh, 2019-01-29:
While running several example sketches (but also on own test sketches) I got this error while compiling in Arduino:
```
In file included from C:\Users\*****\Documents\Arduino\libraries\ArduinoMenu-master\src/menu.h:15:0,

                 from C:\Arduino\Wemos D1 mini\Mini_alarm_ArduinoMenu_test\Mini_alarm_ArduinoMenu_test.ino:11:

C:\Users\*****\Documents\Arduino\libraries\ArduinoMenu-master\src/menuDefs.h:25:28: fatal error: streamFlow.h: No such file or directory

     #include <streamFlow.h>

                            ^

compilation terminated.
```

I can temporarely solve this issue by commenting lines 23 to 46 in menuDefs.h:
```
#if defined(MENU_DEBUG) || defined(MENU_ASYNC)
  #ifndef ARDUINO_STREAMING
    #include <streamFlow.h>
  #else
    #include <Streaming.h>
  #endif
#endif

#if defined(MENU_DEBUG) && defined(TRACE)
  #define trace(x) x
#else
  #define trace(x)
#endif
#ifdef MENU_DEBUG
  #define _trace(x) x
#else
  #define _trace(x)
#endif
#ifndef MENU_RELEASE
  #define __trace(x) x
#else
  #define __trace(x)
#endif
```
But I'm wondering whether there is a better more structural solution.

It's quite like issues #1 and #36 

I did run the sketches on a Wemos D1 mini clone (ESP8266) with U8G2 compatible NONAME 128x64 OLED screen and 3 buttons (1 internal pull up, 2 external pull ups).

neu-rah, 2019-02-02:
solved i guess, please reopen if not

closed 2019-02-02 by neu-rah

Joennuh, 2019-02-02:
Indeed. Resolved by installing both streaming library's. It would be best to mention the need of these library's for ESP8266 devices in the Wiki.