neu-rah/ArduinoMenu #229: examples/adafruitGfx/lcdMono/lcdMono.ino compile error

iZhangHui, 2019-02-09:
Arduino: 1.8.8 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\msi-\Documents\Arduino\libraries\ArduinoMenu_library\examples\adafruitGfx\lcdMono\lcdMono\lcdMono.ino:17:0:

C:\Users\msi-\Documents\Arduino\libraries\ArduinoMenu_library\src/menuIO/encoderIn.h:13:56: fatal error: pcint.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.

PS: @neu-rah Could you give an example how to use Adafruit's GFX devices SSD1306?

d21d3q, 2019-02-09:
Did you notice [comment](https://github.com/neu-rah/ArduinoMenu/blob/f2a71629813c349ba983342536a164a73d70a8ec/src/menuIO/encoderIn.h#L13) in line which crashed your compilation? 

You have to install [this](https://github.com/neu-rah/PCINT) library.

iZhangHui, 2019-02-09:
It works for me now. Thank you! @d21d3q

closed 2019-02-09 by iZhangHui