neu-rah/ArduinoMenu #232: Adding new class TFT_eSPI need some changing

fa1ke5, 2019-02-14:
hi, i'm adding some changing in macros.h for correct support new class "TFT_eSPIOut:"
code example
MENU_OUTPUTS(out, MAX_DEPTH
             //,ADAGFX_OUT(gfx,colors,6*textScale,9*textScale,{0,2,14,8},{14,2,14,8})// old variant
             , TFT_eSPI_OUT(gfx, colors, 6 * textScale, 9 * textScale, {3, 2, 16, 8}) //new class, new variant
             , NONE
            );

neu-rah, 2019-02-14:
Hi! you don't want do a pull request?

fa1ke5, 2019-02-14:
I do it, but I don't it complete or not

neu-rah, 2019-02-14:
yes, do a pull request from your fork (with your changes), git will sort out the differences and will report them on the pull request for approval... so don't worry.

fa1ke5, 2019-02-15:
ok, im try to commit, nice day.

neu-rah, 2019-02-15:
done, thank you :)

closed 2019-02-15 by neu-rah

ddosoff, 2020-10-30:
I search throw the whole project and did not found any example of this macro.

Could you explain parameters of: TFT_eSPI_OUT(gfx, colors, 6 * textScale, 9 * textScale, {3, 2, 16, 8})?

Example will be awesome.

fa1ke5, 2020-10-30:
Насколько я понимаю в текущей версии вы этот код уже не найдете, там видоизменили макросы, проект несколько продвинулся вперед. Я это добавлял когда прикручивал поддержку TFT_eSPI, на тот момент код поддерживал макросы такого формата.

ddosoff, 2020-10-31:
This macro still in the master branch https://github.com/neu-rah/ArduinoMenu/blob/master/src/macros.h#L93

But it's hard to understand preprocessor magic for the user who is unfamiliar with the whole library logic.