neu-rah/ArduinoMenu #235: How to change menuField properties?

ferchinas, 2019-02-16:
Hello: I would like to know if it is possible to change (in execution time) the maximum value that should be allowed. I initialized it in 25.0 but I wanted to be able to change it later.
Thank you

`prompt* mainData[] = {
 new menuField<typeof(floatVal)>(floatVal,"set","",0.0,25.0,0.1,1,(Menu::callback)floatChange,updateEvent),
new Exit("< Exit")
};`

neu-rah, 2019-02-16:
yes, except if you are using an AVR with defaults, for AVR's it needs to be compiled with -`MENU_USERAM` defined at compiler command line.

properties: low,high,step,tune,...

field properties are here https://github.com/neu-rah/ArduinoMenu/blob/master/src/shadows.h#L114

_tech note: shadows.h file contains menu items data that can be static, and put into flash memory, but that is only used by avr's_

ferchinas, 2019-02-17:
It works like a charm!
I'm using an esp32, sorry for not having specified it.
I just had to add the type to the template to compile.

`((menuFieldShadowRaw <float> *) mainData [0] -> shadow) -> high = 50;`

Thank you very much for your help

closed 2019-02-17 by ferchinas

kissgyula, 2019-04-04:
Hello.

I try to do the same, but init the menu by using macros. 

If I try to use this line, I get this error:

```
src\main.cpp:243:40: error: invalid types 'int[int]' for array subscript
((menuFieldShadowRaw <int> *) speed[0]->shadow) -> high=max_speed;
```

Something I missing. Where and how exatly use this?

neu-rah, 2019-04-04:
is it an `int` field?
on that line, you are setting the range high value to `max_speed`

it is only possible to do so on non-AVR devices or if compiling with special defines at compiler cmd line...

assuming that **speed** is a menu and that the first element [0] is an **int** field

kissgyula, 2019-04-04:
I'm also using an esp32.

I think it should be an int. I using the variables as an int.

Menu item is definied in this line:
`,FIELD(speed,"Speed:","",0,max_speed,100,1,setMotorSpeed,exitEvent,wrapStyle)`

I thought "speed" should be the name of the menu item. But i is the third element of the menu mainMenu. I modified the line:
`((menuFieldShadowRaw <int> *) mainMenu[2]->shadow) -> high=max_speed;`

Now the error is:
```
src\main.cpp: In function 'void setMotorMaxSpeed()':
src\main.cpp:243:33: error: 'mainMenu' was not declared in this scope
((menuFieldShadowRaw <int> *) mainMenu[2]->shadow) -> high=max_speed;
```
Is there some way to predefine the menu?

neu-rah, 2019-04-04:
`mainMenu` not defined yet, guess its defined latter

insert a prototype of your `setMotorMaxSpeed` and add its implementation after `mainMenu` is defined (assuming that the field is on some menu called mainMenu :D )

kissgyula, 2019-04-04:
Ok. Its working now.

Thanks for the quick help.