neu-rah/ArduinoMenu #236: How to synchronize variables

ferchinas, 2019-02-19:
I am working on an ESP32 with an Oled i2c display.

```
bool		menuVar;
prompt* togData[] = {
	new menuValue<bool>("On",true),
	new menuValue<bool>("Off",false)
};
toggle<bool>& toggleField = *new toggle<bool>("Auto ", menuVar, sizeof(togData) / sizeof(prompt*), togData, (Menu::callback)enter_cb, enterEvent);
loop(){
	if (nav.changed(0)) {
		getValues();				
		Display.firstPage();
		do nav.doOutput(); while (Display.nextPage());
	}
}
getValues(){
	menuVar = myObject.Var;
}
void	enter_cb(eventMask e) {
	myObject.Var = menuVar;
}
```

I need to modify in AruinoMenu member of objects "myObject.Var" (created at runtime).

I can reflect the changes made in ArduinoMenu with the function "enter_cb (eventMask e)"

I also need to reflect in "menuVar" the changes I make to "myObject.Var" outside the menu. With "getValues ()" I can update "menuField" of type float, but with variables of type "bool" I can not see the changes reflected in "myObject.Var"

What is the way to update the variables used by ArduinoMenu?
Thank you

neu-rah, 2019-02-22:
ArduinoMenu does not store values/variables (except for backups) instead it hooks to your vars by reference, so any changes you do on menu interface are automatically done in your vars and changes your code does to your vars are reflected on menu interface with no extra code (that's why it stores a backup.

Menu only checks for changes on visible options and only redraws if values changed (with some exceptions on some output devices).

lorol, 2019-02-23:
@ferchinas You can try this (easy way):
bool isauto;

TOGGLE(isauto, autoMenu, "Auto", doNothing, noEvent, noStyle
       , VALUE("On", false, do_on, enterEvent)
       , VALUE("Off", true, do_off, enterEvent)
      );
...
MENU(mainMenu, " Title ", doNothing, noEvent, noStyle
     , SUBMENU(autoMenu)
...

result do_on() { //when you toggle menu to "On" it will change isauto to true and call this 
 ...
  return proceed;
}

somewhere else:  isauto = false; will change the display as well.

neu-rah, 2019-02-23:
ahh, I missed that **Created at runtime part**, sorry was rushing

you can use the events for that (as you did), by replacing the `doNothing` by your handler and setting the proper event mask.
Reflecting to screen, its just a matter of updating the intermediary variable, the menu will notice the change. You can also force the drawing of menu items by setting `dirty=true` (ex: `toggleField.dirty=true`), but in this case, it's not needed because we have a variable.

Am I missing any important detail this time? If so please call my attention to it, because I'm rushing again :D

ferchinas, 2019-02-25:
Perfect! I can now synchronize the variables properly. Thank you!

closed 2019-02-25 by ferchinas