neu-rah/ArduinoMenu #237: Custom menuFlied to select float decimal places

ferchinas, 2019-02-20:
Hi Rui:
I made the custom menuField to work, at least for my setup
ESP32 "core:1.0.0"
HW Oled i2C 0.96
u8g2 driver

The number of places for decimals is defined by the user, you can change it at any time of the execution.
Each field can have a different number of decimal places.

I share an example for those who may need it

```
#include <menu.h>
#include <menuIO/serialOut.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialIn.h>

using namespace Menu;

#define MAX_DEPTH 1
unsigned char cantDecimals = 1;
float floatVar = 10.123456;

void updateEvent_cb(eventMask e);

//-----Custom floatField----------------
#define FLOATFLIED_DECIMALS_DEFAULT 1

template<typename T>
class floatlField :public menuField<T> {                
private:
  idx_t decimals;
public:
  floatlField(constMEM menuFieldShadow<T>& shadow) :menuField<T>(shadow) { decimals = FLOATFLIED_DECIMALS_DEFAULT; }
  floatlField(
    T &value,
    constText* text,
    constText*units,
    T low,
    T high,
    T step,
    T tune,
    action a = doNothing,
    eventMask e = noEvent,
    styles s = noStyle
  ) :floatlField(*new menuFieldShadow<T>(value, text, units, low, high, step, tune, a, e, s)) {}

  Used printTo(navRoot &root, bool sel, menuOut& out, idx_t idx, idx_t len, idx_t panelNr = 0) override {
    //menuFieldShadow<T>& s=*(menuFieldShadow<T>*)shadow;
    menuField<T>::reflex = menuField<T>::target();
    idx_t l = prompt::printTo(root, sel, out, idx, len);
    bool ed = this == root.navFocus;
    //bool sel=nav.sel==i;
    if (l < len) {
      out.print((root.navFocus == this&&sel) ? (menuField<T>::tunning ? '>' : ':') : ' ');
      l++;
      if (l < len) {
        l += out.print(menuField<T>::reflex, decimals);//NOTE: this can exceed the limits!
        if (l < len) {
          l += print_P(out, fieldBase::units(), len);
        }
      }
    }
    return l;
  }

  void setDecimals(idx_t d) { decimals = d; }
  idx_t getDecimals(void) { return(decimals); }
};
//-----Custom floatField----------------END

 
floatlField <typeof(floatVar)>& floatFlied_Object = *new floatlField<typeof(floatVar)>(floatVar, "Min", "", 0.0, 100.0, 0.1, 1);
floatlField <typeof(cantDecimals)>& DecimalsFlied_Object = *new floatlField<typeof(cantDecimals)>(cantDecimals, "Dec", "", 0, 6, 1, 1, (Menu::callback)updateEvent_cb, updateEvent);

void updateEvent_cb(eventMask e){
  floatFlied_Object.setDecimals(cantDecimals);
  Serial.println("");Serial.print("now using '");Serial.print(cantDecimals);Serial.println("' decimals.");
 }
 
//---Main menu  ---------------------------------
prompt* mainData[] = {
  &floatFlied_Object,
  &DecimalsFlied_Object,
  new Exit("<Back")
};

menuNode& mainMenu = *new menuNode("Main menu", sizeof(mainData) / sizeof(prompt*), mainData);

serialIn serial(Serial);
MENU_INPUTS(in,&serial);

MENU_OUTPUTS(out,MAX_DEPTH
  ,SERIAL_OUT(Serial)
  ,NONE//must have 2 items at least
);

NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);

void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial.println("Menu 4.x");
  Serial.println("Use keys + - * /");
  Serial.println("to control the menu navigation");
  nav.useUpdateEvent = true;
}

void loop() {
  nav.poll();
}

```

neu-rah, 2019-02-21:
excellent!
don't you want to do a pull request? putting this at `plugin` folder would be very nice even if specific to some setup (plugins can be specific). However, this one seems quite generic.

ferchinas, 2019-02-21:
Made [PR](https://github.com/neu-rah/ArduinoMenu/pull/238/commits/3ca99c10630e096e831420884c49badce11a5570)
I'm not an expert, suggestions are welcome.

closed 2019-02-21 by ferchinas

Bflorin-sk, 2020-02-20:
how to make this work on arduino? its not displaying properly for me on serial monitor of IDE.

neu-rah, 2020-02-20:
are you using the floatlField above?

Bflorin-sk, 2020-02-20:
i am using everything as it is. and i tried to uploadit to a arduino UNO and evrithing coompiles smooth no erors
but ..
11:50:30.219 -> Menu 4.x
11:50:30.219 -> Use keys + - * /
11:50:30.219 -> to control the menu navigation
11:50:30.219 -> ⸮⸮⸮⸮⸮⸮6⸮%⸮4⸮&⸮⸮⸮⸮%⸮4⸮⸮⸮⸮⸮⸮ߑϑ⸮

neu-rah, 2020-02-20:
original using esp32, guess there is some discrepancies over memory allocation RAM vs FLASH

what IDE are you using?

Bflorin-sk, 2020-02-20:
1.8.12

neu-rah, 2020-02-20:
arduino IDE wont let you specify compiling flash (at least last time i checked) otherwise we could force everything to RAM to make a quick check... I'm quite sure this example needs some review to work on flash memory, adding to my todo list

Bflorin-sk, 2020-02-20:
May 2017 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
www.r-site.net
unlike native objects, pluggins can be device specific, so some pluggins might
not work on all devices.
all sorts of plugins may be available in the future and if you customize
a component and think it of interest of others please do pull request.
contrubutions and contrubutors are welcome.
decimalslField - To select the number of decimal places that are printed in float and double variables types. Feb 2019 ferchinas
                 Tested on ESP32, Oled i2c display, u8g2 driver. It could work on other devices.
***/     

this is on the beginning of the example

neu-rah, 2020-02-20:
indeed, as a plugin it would work for all mcu's were menu uses ram by default, not the case of AVR's

Bflorin-sk, 2020-02-20:
ok as a possible improvement it is possible to make a normal expression (123.456...) of decimal number followed by a multiplier/denominator like Mega ,kilo, mili, nano.... (  123.45 [kilo])

neu-rah, 2020-02-20:
there is also an `unit` text available for all numeric fields, however its not dynamic

as a note, improvements like this will probably be added as plugins because existing features are already tight on tiny mcu's like atmega328

thanks for the feed back, its really appreciated

Bflorin-sk, 2020-02-20:
for my case there should be no trouble because i intend to to use a ST7920 12864 display with a rotary encoder and Arduino mega as a  HMI. the uno was used only for testing stuff that i intend to use.

neu-rah, 2020-02-20:
meanwhile and if changing decimal places for all numeric fields is ok for you, you can change code at https://github.com/neu-rah/ArduinoMenu/blob/master/src/itemsTemplates.hpp#L37

to:
```c++
  template<typename T>
  idx_t menuField<T>::printReflex(menuOut& o) const {return o.print(reflex,3);}
```

here using 3 decimal places

neu-rah, 2020-02-20:
Hi!

added a decimal places format field, not with a fancy unit scale print (like Mega ,kilo, mili, nano....) but allowing only specification of decimal places, was do simple that no plugin added, now its part of the lib

usage example here https://github.com/neu-rah/ArduinoMenu/blob/master/examples/handlers/handlers/handlers.ino#L140

```c++
altFIELD(decPlaces<3>::menuField,test,"Test","%",0,100,1,0.001,showEvent,anyEvent,wrapStyle)
```

hope it works ok, let me know if not

Bflorin-sk, 2020-02-20:
17:27:01.853 -> [-] Alert test
17:27:01.853 -> [-] Name                                
17:27:01.853 -> [-] Hex 0x11
17:27:03.978 -> ========
17:27:03.978 -> Event for target: 0x55A
17:27:03.978 -> nav level:0 path:[3]
17:27:03.978 -> 128 selBlurEvent
17:27:03.978 -> 
17:27:03.978 -> ========
17:27:03.978 -> Event for target: 0x55A
17:27:03.978 -> nav level:0 path:[3]
17:27:03.978 -> 32 blurEvent
17:27:03.978 -> 
17:27:03.978 -> ========
17:27:03.978 -> Event for target: 0x55A
17:27:03.978 -> nav level:0 path:[2]
17:27:03.978 -> 16 focusEvent
17:27:03.978 -> 
17:27:03.978 -> ========
17:27:03.978 -> Event for target: 0x55A
17:27:03.978 -> nav level:0 path:[2]
17:27:03.978 -> 64 selFocusEvent
17:27:04.013 -> 
17:27:04.013 -> 
17:27:04.013 -> Main menu
17:27:04.013 -> [3]>Test 55.000%
17:27:04.013 -> [4] Sub-Menu
17:27:04.013 -> [5] Led: Off
17:27:04.013 -> [6] LED On
17:27:04.013 -> [7] LED Off
17:27:04.013 -> [8] Select Zero
17:27:04.013 -> [9] Choose Last
17:27:04.013 -> [-] Alert test
17:27:04.013 -> [-] Name                                
17:27:08.029 -> ========
17:27:08.029 -> Event for target: 0x55A
17:27:08.029 -> nav level:0 path:[2]
17:27:08.029 -> 4 exitEvent
17:27:08.029 -> 
17:27:08.029 -> suspending menu!
17:27:08.029 -> suspended...

Bflorin-sk, 2020-02-20:
ok i testedit on my device Mega +encoder+12864 and its wierd

Bflorin-sk, 2020-02-20:
it differs from the example i used were first i set the decade and then the units
this example setts everithing before the "," by counting with the encoder then everyting the same after  the ",".

Bflorin-sk, 2020-02-20:
`

#include <Arduino.h>

/********************
Arduino generic menu system
U8G2 menu example
U8G2: https://github.com/olikraus/u8g2
Oct. 2016 Stephen Denne https://github.com/datacute
Based on example from Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
Original from: https://github.com/christophepersoz
menu on U8G2 device
output: Wemos D1 mini OLED Shield (SSD1306 64x48 I2C) + Serial
input: Serial + encoder
mcu: nano328p
*/

#include <menu.h>  //menu macros and objects
#include <menuIO/u8g2Out.h>
#include <TimerOne.h>
#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h> 
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h> // concatenate multiple input streams (this allows adding a button to the encoder)
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>

using namespace Menu;

#define LEDPIN LED_BUILTIN

// #define USE_PCD8544
//#define USE_SSD1306

//#define(USE_PCD8544)
  // rotary encoder pins
   #define encA    A0
   #define encB    A1
   #define encBtn  A2
   #define encSteps 2

ClickEncoder clickEncoder = ClickEncoder(encA, encB, encBtn, encSteps);
ClickEncoderStream encStream(clickEncoder, 1);
//ClickEncoder clickEncoder(encA,encB,encBtn,2);
//ClickEncoderStream encStream(clickEncoder,1);
//MENU_INPUTS(in,&encStream);

void timerIsr() {clickEncoder.service();}
result doAlert(eventMask e, prompt &item);
result showEvent(eventMask e,navNode& nav,prompt& item) {
  Serial.print("event: ");
  Serial.println(e);
  return proceed;
}

//#define U8G2_16BIT///////////
  #include <SPI.h>
  #define USE_HWSPI
 // #define U8_DC 9
 // #define U8_CS 10
 // #define U8_RST 8
  #define fontName u8g2_font_6x10_tr
  //#define fontName u8g2_font_chikita_tr

  #define fontX 6
  #define fontY 12
  #define offsetX 0
  #define offsetY 0
  #define U8_Width 128
  #define U8_Height 64
[... 238 lines cut ...]
    //u8g2.setFont(u8g2_font_unifont_t_symbols);
    
    //u8g2.setCursor(2+test, 32);
     u8g2.setFont(u8g2_font_4x6_tr);
    u8g2.drawStr(test-1, 64,"^"); ///this should become setpoint
    u8g2.setCursor(2+test , 64);
    u8g2.print(test,3); 
        
   u8g2.setFont(u8g2_font_6x10_tr);
    } while ( u8g2.nextPage() );
  } else {
    if (nav.changed(0)) {
      
      u8g2.firstPage();
      
      do nav.doOutput(); while(u8g2.nextPage());
    }
  }
   nav.timeOut=300;  //menu goes idle after 300seconds of not useage
  delay(25);//simulate other tasks delay
    }
    
//}

//}`

Bflorin-sk, 2020-02-20:
#include <Arduino.h>

/********************
Arduino generic menu system
U8G2 menu example
U8G2: https://github.com/olikraus/u8g2
Oct. 2016 Stephen Denne https://github.com/datacute
Based on example from Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
Original from: https://github.com/christophepersoz
menu on U8G2 device
output: Wemos D1 mini OLED Shield (SSD1306 64x48 I2C) + Serial
input: Serial + encoder
mcu: nano328p
*/

#include <menu.h>  //menu macros and objects
#include <menuIO/u8g2Out.h>
#include <TimerOne.h>
#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h> 
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h> // concatenate multiple input streams (this allows adding a button to the encoder)
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>

using namespace Menu;

#define LEDPIN LED_BUILTIN

// #define USE_PCD8544
//#define USE_SSD1306

//#define(USE_PCD8544)
  // rotary encoder pins
   #define encA    A0
   #define encB    A1
   #define encBtn  A2
   #define encSteps 2

ClickEncoder clickEncoder = ClickEncoder(encA, encB, encBtn, encSteps);
ClickEncoderStream encStream(clickEncoder, 1);
//ClickEncoder clickEncoder(encA,encB,encBtn,2);
//ClickEncoderStream encStream(clickEncoder,1);
//MENU_INPUTS(in,&encStream);

void timerIsr() {clickEncoder.service();}
result doAlert(eventMask e, prompt &item);
result showEvent(eventMask e,navNode& nav,prompt& item) {
  Serial.print("event: ");
  Serial.println(e);
  return proceed;
}

//#define U8G2_16BIT///////////
  #include <SPI.h>
  #define USE_HWSPI
 // #define U8_DC 9
 // #define U8_CS 10
 // #define U8_RST 8
  #define fontName u8g2_font_6x10_tr
  //#define fontName u8g2_font_chikita_tr

  #define fontX 6
  #define fontY 12
  #define offsetX 0
  #define offsetY 0
  #define U8_Width 128
  #define U8_Height 64
  
  //U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0, /* CS=*/ 10, /* reset=*/ 8);
[... 236 lines cut ...]
    //u8g2.setFont(u8g2_font_unifont_t_symbols);
    
    //u8g2.setCursor(2+test, 32);
     u8g2.setFont(u8g2_font_4x6_tr);
    u8g2.drawStr(test-1, 64,"^"); ///this should become setpoint
    u8g2.setCursor(2+test , 64);
    u8g2.print(test,3); 
        
   u8g2.setFont(u8g2_font_6x10_tr);
    } while ( u8g2.nextPage() );
  } else {
    if (nav.changed(0)) {
      
      u8g2.firstPage();
      
      do nav.doOutput(); while(u8g2.nextPage());
    }
  }
   nav.timeOut=300;  //menu goes idle after 300seconds of not useage
  delay(25);//simulate other tasks delay
    }
    
//}

//}

Bflorin-sk, 2020-02-20:
ok its very nice actually and it works great (my mistake forgot to change from int test to float test) but.... i have no control of the amount of zeroes after ",". or do I?
or somthing like i write(1234567) and then move the "," where i need it like (1,234567),(12,34567),(123,4567) as means of inputing a value,
or just something like 1.2345*10^3 scientific calculator format.

neu-rah, 2020-02-20:
yes you do, on this part `decPlaces<3>` change the number to desired places

neu-rah, 2020-02-20:
the implementation is quite trivial, it took me some time to find the better way

```c++
      template<uint8_t dps>//here a template parameter for the number of places
      struct decPlaces {
        template<typename T>
        struct menuField:Menu::menuField<T> {//and here we are reusing the old menuField class
          using Menu::menuField<T>::menuField;//even reusing the constructor as extra parameter are on outside template
          idx_t printReflex(menuOut& o) const override;//only to change the way it prints out
        };
      };
```

and the we can implement the printing to whatever we want

```c++
  template<uint8_t dps>
  template<typename T>
  idx_t decPlaces<dps>::menuField<T>::printReflex(menuOut& o) const {
    return o.print(Menu::menuField<T>::reflex,dps);//just by changing here what its printed
  }
```

Bflorin-sk, 2020-02-20:
` //,altFIELD(decPlaces<3>::menuField,test,"Test","%",0,100,1,0.001,showEvent,anyEvent,wrapStyle)//works ok/
,altFIELD(decPlaces<4>::menuField,test,"Test","%",0,100,1,0.0001,showEvent,anyEvent,wrapStyle)//works ok///`  it works

Bflorin-sk, 2020-02-20:
but... still...   lets say in my case i can only change value by  selecting befor or after "'," and incrementing (counting by encoder ) to the desired value if i have 55.0000 and want to change it to 
55.4321 it is a lot of counting by manual encoder so ...
i was thinking of something like this
>55.4000
>55.4300
..
>55.4321
{confirm by pressing ... stuff... encButton}.

neu-rah, 2020-02-20:
if you are using serial input you can type in the value for the field (nor checked with floats)...

clickencoder can use acceleration, rotating to same direction increases the stepping

Bflorin-sk, 2020-02-20:
something like
`,FIELD(test,"Test","%",0,100,10,1,doNothing,noEvent,wrapStyle`

Bflorin-sk, 2020-02-20:
off coures here are only 2 digits

neu-rah, 2020-02-20:
you're welcome to join us at https://gitter.im/ArduinoMenu/Lobby

Bflorin-sk, 2020-02-20:
> if you are using serial input you can type in the value for the field (nor checked with floats)...
> 
> clickencoder can use acceleration, rotating to same direction increases the stepping

it seems i cant get acceleration working at first i thought my hardware debouncer is overkill . but..
i changed it and the same
whatever speed i turn the encoder it does not accelerate (change rate)

neu-rah, 2020-02-20:
are you using the clickencoder (https://github.com/0xPIT/encoder)?

if not working please open a new issue, we are getting off topic here