neu-rah/ArduinoMenu #239: Strange behavior with ATSAMD21 Cortex M0

profra, 2019-03-09:
Trying to use ArduinoMenu with board Adafruit ItsyBitsy M0 Express (framework = arduino in Platformio)...
It seems that the input characters are treated incorrectly... somehow delayed or lost. So it means that this library is unusable in this environment... It would be a pity. Can you fix it somehow?

neu-rah, 2019-03-09:
it runs ok on Arduino due,  Atmel SAM3X8E ARM Cortex-M3 CPU.
what example are you using?
what IDE, Arduino_IDE?

profra, 2019-03-09:
I tried the first example, Blink... IDE is Platformio...
here is platformio.ini...
```
[env:adafruit_itsybitsy_m0]
platform = atmelsam
board = adafruit_itsybitsy_m0
framework = arduino

monitor_speed = 115200
upload_protocol = sam-ba

lib_deps =
  ArduinoMenu library

;WIN
;upload_port = COM32

;NIX
;upload_port = /dev/ttyACM0
```

neu-rah, 2019-03-09:
please try Serial example instead, 
it's my fault, I gotta make a new blink example, one that does not use delay because that delay can make the system sluggish.

using platformio too :+1:

profra, 2019-03-09:
I did a test with "serialio" and it seems that it works. I have to go through it because I do not quite understand every detail. It was a shame that I chose Blink for the first test... but it is logical if someone wants to do a fast test with a new thing :-). It would be good to make a new BLINK otherwise it is a trap for new enthusiasts :-)
I agree platformio is excellent :+1:

neu-rah, 2019-03-09:
thanks for the interest, shame on me of such a bad example
there is also a chat room about this menu, https://gitter.im/ArduinoMenu/Lobby
feel free to ask there too.

closed 2019-03-09 by neu-rah

profra, 2019-03-09:
Thanks for info about chat room... I will visit it for sure!