neu-rah/ArduinoMenu #241: sdmenu and u8g2 rendering seems wrong

pcharest2000, 2019-03-19:
Platform:  Arduino bluebill
My menus are rendered properply but when using the sdmenu plugin the background of certain items are not renderd properly.  I have tried both 
u8g2_font_profont12_mr  and u8g2_font_profont12_tr fonts with diffrent results... please find links to images
In the SD menu using the tr font the unselected text seems partially rendred (some background kept),  using the mr font  the whole background of the text is rendred white.

I
[image]
[image]
[image]
[image]

Tks for your help and great work

Phil

neu-rah, 2019-03-19:
thanks for your feedback, images gave me a much more clear picture, I'm not using SDMenu on my projects yet, it's recent and I only tested on serial.
Had one report of this, but now it's clear :+1: 
will give it a try on some u8g2 display, btw, what display is that?

pcharest2000, 2019-03-20:
The display is à 128x64 st7920 using the spi bus.  Ive also tried. Turnising on and of setbitmapmode of u8g2 on and off to no effect.

neu-rah, 2019-07-21:
this should solve it, reopen if not

closed 2019-07-21 by neu-rah

hicksan, 2021-08-02:
> this should solve it, reopen if not

How was this solved please? I cannot get this function to work, even without a display. The example SDCard.Ino file, without modification, fails during Setup. The standard SD card library examples all work perfectly, with the same hardware setup. Has something in menu.h been modified in recent varsions so it no longer works?