neu-rah/ArduinoMenu #244: Main screen with menu activated on click

woodencase01, 2019-06-15:
Hi!

I'm building a menu system for a benchmarking tool which has many sensors.
I'd like to have the main screen showing all the sensors current values and have the menus appear on my encoder clicks.

I'm having difficulties figuring out how to do it.

I'd like to show all sensors such as below on the main screen:
`|-CURRENT STATE-|`
`|-TEMP 1--TEMP 2--TEMP 3--|`
`|-POS 1---POS 2---POS 3---|`
`|-SPEED 1-SPEED 2-SPEED 3-|`

What's the best way to achieve this?

Here is my current code:
```
void readButtons()
{
    clickEncoder.service();
}

const colorDef<uint8_t> colors[] MEMMODE={
  {{0,0},{0,1,1}},//bgColor
  {{1,1},{1,0,0}},//fgColor
  {{1,1},{1,0,0}},//valColor
  {{1,1},{1,0,0}},//unitColor
  {{0,1},{0,0,1}},//cursorColor
  {{1,1},{1,0,0}},//titleColor
};

bool heatingState;

result menuGetHeatingState()
{
    heatingState = getHeatingState();
    return proceed;
}

result menuTurnOnHeaters()
{
    heatingOn();
    return proceed;
}

result menuTurnOffHeaters()
{
    heatingOff();
    return proceed;
}

TOGGLE(heatingState,toggleHeatersMenu,"Heaters are: ",menuGetHeatingState,noEvent,wrapStyle
  ,VALUE("On",HIGH,heatingOn,noEvent)
  ,VALUE("Off",LOW,heatingOff,noEvent)
);

extern float global_extrusionEfficiency;

CHOOSE(global_extrusionEfficiency,efficiencyMenu,"Efficiency: ",doNothing,noEvent,noStyle
  ,VALUE("Rigid",0.75f,doNothing,noEvent)
  ,VALUE("Flexible",0.50f,doNothing,noEvent)
);

MENU(mainMenu,"Benchmark Tool",doNothing,noEvent,wrapStyle
    ,EXIT("<Exit")
    ,SUBMENU(toggleHeatersMenu)
    ,SUBMENU(efficiencyMenu)
  );

const int MAX_DEPTH = 2;

MENU_OUTPUTS(out,MAX_DEPTH
  ,U8G2_OUT(u8g2,colors,fontX,fontY,offsetX,offsetY,{0,0,U8_Width/fontX,U8_Height/fontY})
  ,NONE
);

NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);
```

neu-rah, 2019-06-15:
Hi!

on setup:
```c++
  nav.idleTask=myIdleScr;//point a function to be called when menu exists
  nav.timeOut=60;//optionally define a timeout in seconds
  nav.idleOn(myIdleScr);//optionally start the system with the idle screen
```

the menu will idle either by timeout or by exiting the main menu and resume on select/exit command or programaticallly with `nav.idleOff()`

example of idle function, put it before setup
```c++
//the idle handler
result myIdleScr(menuOut& o,idleEvent e) {
  switch(e) {
    case idleStart:
      //enter idle
      //can be called multiple times, depending of the number of output devices
      break;
    case idleEnd:
      //terminate idle, can be called multiple times, depending on output device type
      break;
    case idling:
      //do idle stuff
      nav.idleChanged=true;//force redraw of idle screen (we need to update content)
      break;
  }
  return proceed;
}
```

i've not compiled this, just copy paste from existing code, so let me know if i skipped or miss renamed some stuff.

there is some info on "idling" here https://github.com/neu-rah/ArduinoMenu/wiki/Idling

thanks for the interest.

neu-rah, 2019-06-17:
feel free to reopen if not solved or if you need more info

closed 2019-06-17 by neu-rah