neu-rah/ArduinoMenu #245: example esp8266/WebMenu do not compile

pat1, 2019-06-17:
esp8266 2.5.2
arduino 1.8.5

error:
```
"/home/pat1/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/pat1/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include" "-I/home/pat1/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include" "-I/home/pat1/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_668036/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINIPRO -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINIPRO"  -DFLASHMODE_DIO  -DESP8266 "-I/home/pat1/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266" "-I/home/pat1/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/variants/d1_mini" "-I/home/pat1/git/rmap/arduino/sketchbook/libraries/ArduinoMenu/src" "-I/home/pat1/git/rmap/arduino/sketchbook/libraries/streamFlow/src" "-I/home/pat1/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src" "-I/home/pat1/git/rmap/arduino/sketchbook/libraries/WebSockets/src" "-I/home/pat1/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src" "-I/home/pat1/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/Hash/src" "/tmp/arduino_build_668036/sketch/WebMenu.ino.cpp" -o "/tmp/arduino_build_668036/sketch/WebMenu.ino.cpp.o"
WebMenu:218: error: 'len' declared as an 'inline' variable
 constexpr inline size_t len(T& o) {return sizeof(o)/sizeof(decltype(o[0]));}
                             ^
WebMenu:218: error: 'T' was not declared in this scope
WebMenu:218: error: 'o' was not declared in this scope
 constexpr inline size_t len(T& o) {return sizeof(o)/sizeof(decltype(o[0]));}
                                ^
WebMenu:218: error: 'template<class T> constexpr size_t len(T&)' redeclared as different kind of symbol
 constexpr inline size_t len(T& o) {return sizeof(o)/sizeof(decltype(o[0]));}
                                 ^
WebMenu:218: error: previous declaration of 'constexpr const size_t len'
 constexpr inline size_t len(T& o) {return sizeof(o)/sizeof(decltype(o[0]));}
```

neu-rah, 2019-06-17:
there seem to be a name collision... i'm renaming the function on that example as it is only used to calculate the menu data length, if problems persist please let me know because the original and the changes are both compiling ok on my environment.

neu-rah, 2019-06-22:
please reopen if not solve

closed 2019-06-22 by neu-rah