neu-rah/ArduinoMenu #246: not implemented "Scrollbar" capability

HamidSaffari, 2019-07-11:
Hi and big bow to the the creators of this great and complicated library.
Would you implement Scrollbar capability for not fitted fonts and menus?

HamidSaffari, 2019-07-11:
like this one from:
https://github.com/Jomelo/LCDMenuLib2

[image]

neu-rah, 2019-07-12:
Hi! Thanks for your interest

this can also be implemented as a plugin... and why plugins? because fancy features grow the code base and this library is almost impossible to use with a graphics library on a small mcu like atmega328.

This feature requires a gfx library or special chars on a text output device. So it fits the plugin place.

this can be implemented aside the menu central code, just a matter of reducing the panel size to let space for drawing the scroll bar and draw it according to menu proportions and navigation position.

about the `X` cursor, its a setting for text output but graphics output will ignore it and use a selectin cursor (inverted graphics) instead.