neu-rah/ArduinoMenu #249: Structure elements

Ulli2k, 2019-07-12:
Is it possible to add a header / structure elements? 
I want to have it look like
menu header
------
field
submenu
....
------
menu footer

neu-rah, 2019-07-12:
yes, that's the role of this library, what board/mcu and IO are you using?

code examples here: https://github.com/neu-rah/ArduinoMenu/tree/master/examples

videos on main page also menu examples.

Ulli2k, 2019-07-13:
I am using a ESP32.
I already saw the examples. But I did not find å way to add headlines.

neu-rah, 2019-07-13:
ahh, ok
you can limit the area of the menu on the output device, by changing the panel coordinates... trimming the right side of text might be disabled on some devices (like u8g2 displays) so, one has ro make sure text fits the  space.

- footer, some space left for footer (lines limit are always respected on all device) one has some area to free draw... we can coordinate that with the active menu, by using the events... as for the text source to figure on footer it can be either external or internal if we derive the existent menu objects to support an extra text

- separators, we can use disabled options, they will however take a whole line.  
drawing graphics lines between the option seems complicated and not all devices support

a good principle to face this issues would be variable height options, current version uses only lines
and work with device coordinates instead of text grids as the current menu version does

as a general principle things that are hardware specific or require some special capabilities are implemented as plugins, so that the base objects are supported by all devices including simple text output (monitor, console, printer).

developement discussions also here: https://gitter.im/ArduinoMenu/Lobby

thanks for the interest
https://gitter.im/ArduinoMenu/Lobby