neu-rah/ArduinoMenu #250: a lot of compiler Warnings

dremeier, 2019-07-14:
I´m just tried to compile some of your examples. But I got nearly always the same Warnings.

Serialio:
`C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\examples\Serial\serialio\serialio\serialio.ino:104:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 char* constMEM hexDigit MEMMODE="0123456789ABCDEF";

                                 ^

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\examples\Serial\serialio\serialio\serialio.ino:105:58: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 char* constMEM hexNr[] MEMMODE={"0","x",hexDigit,hexDigit};

                                                          ^

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\examples\Serial\serialio\serialio\serialio.ino:105:58: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\examples\Serial\serialio\serialio\serialio.ino:108:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 char* constMEM alphaNum MEMMODE=" 0123456789.ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz,\\|!\"#$%&/()=?~*^+-{}[]€";

                                 ^

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\src\itemsTemplates.cpp:1:9: warning: #pragma once in main file [enabled by default]

 #pragma once

         ^

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\src\items.cpp:463:11: warning: extra tokens at end of #endif directive [enabled by default]

     #endif-

           ^

Der Sketch verwendet 28068 Bytes (5%) des Programmspeicherplatzes. Das Maximum sind 524288 Bytes.
`
and Errors on u8g2:
`In file included from C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\src/menuIO/encoderIn.h:13:0,

                 from C:\share\Elektronik\Elektronik-devices-data\Hi-Fi\ICEpower\DAC-OpAmp-PowerSupply\Display_Test\U8G2_OLED_128X64_Test\U8G2_OLED_128X64_Test.ino:21:

C:\Users\Andre\Documents\Arduino\libraries\PCINT-master\src/pcint.h:48:5: warning: #warning using maps! [-Wcpp]

    #warning using maps!

     ^

C:\share\Elektronik\Elektronik-devices-data\Hi-Fi\ICEpower\DAC-OpAmp-PowerSupply\Display_Test\U8G2_OLED_128X64_Test\U8G2_OLED_128X64_Test.ino:152:33: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

 char* constMEM hexDigit MEMMODE="0123456789ABCDEF";

                                 ^

C:\share\Elektronik\Elektronik-devices-data\Hi-Fi\ICEpower\DAC-OpAmp-PowerSupply\Display_Test\U8G2_OLED_128X64_Test\U8G2_OLED_128X64_Test.ino:153:58: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

 char* constMEM hexNr[] MEMMODE={"0","x",hexDigit,hexDigit};

                                                          ^

C:\share\Elektronik\Elektronik-devices-data\Hi-Fi\ICEpower\DAC-OpAmp-PowerSupply\Display_Test\U8G2_OLED_128X64_Test\U8G2_OLED_128X64_Test.ino:153:58: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\src\itemsTemplates.cpp:1:9: warning: #pragma once in main file

 #pragma once

         ^

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\src\items.cpp:463:11: warning: extra tokens at end of #endif directive [-Wendif-labels]

     #endif-

           ^

In file included from C:\Users\Andre\Documents\Arduino\libraries\PCINT-master\src\pcint.cpp:1:0:

C:\Users\Andre\Documents\Arduino\libraries\PCINT-master\src\pcint.h:48:5: warning: #warning using maps! [-Wcpp]

    #warning using maps!

     ^

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\src/itemsTemplates.cpp: In member function 'clearChanged':

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\src/itemsTemplates.cpp:29:3: internal compiler error: Segmentation fault

   }

   ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.`

Can you tell me why there are so many warnings and errors on the default examples?

neu-rah, 2019-07-14:
Hi!

here what i have when compiling `serialio` example ffor arduino nano using IDE 1.8.9

```c++
AVISO: Pasta .pio espúria na biblioteca «ArduinoMenu library»
O rascunho usa 21110 bytes (68%) do espaço de armazenamento do programa. O máximo é 30720 bytes.
Variáveis globais usam 1256 bytes (61%) de memória dinâmica, restando 792 bytes para variáveis locais. O maximo é 2048 bytes.
```

note that I have compilation details turned off on IDE preferences.

what is your target board?

dremeier, 2019-07-14:
Hello, and Thanks for your quick support.
My target board will be the Arduino DUE.
My testing board is the Arduino Mega2560, on the Mega2560 I just got warnings on the Due errors.

```
C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\src/macros.h:193:3: error: cannot convert 'time_t (*)(time_t*) {aka long int (*)(long int*)}' to 'Menu::prompt*' in initialization

   };\

C:\Users\Andre\Documents\Arduino\libraries\ArduinoMenu_library\src/macros.h:186:42: note: in expansion of macro 'altMENU'

 #define MENU(id,text,aFn,mask,style,...) altMENU(Menu::menu,id,text,aFn,mask,style,(Menu::_menuData|Menu::_canNav),__VA_ARGS__)

C:\share\Elektronik\Elektronik-devices-data\Hi-Fi\ICEpower\DAC-OpAmp-PowerSupply\Display_Test\U8G2_OLED_128X64_Test\U8G2_OLED_128X64_Test.ino:156:1: note: in expansion of macro 'MENU'

 MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle

exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Programming Port).
```

exocode, 2019-07-19:
Have the same stacktrace. On a ESP32.
I use PlatformIO.
Here is the complete Console output:
Hope this helps to solve the issue:
````> Executing task: platformio run <

Processing heltec_wifi_kit_32 (platform: espressif32; board: heltec_wifi_kit_32; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_kit_32.html
PLATFORM: Espressif 32 1.9.0 > Heltec WiFi Kit 32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib> 1.19.1
|-- <ArduinoMenu>
|   |-- <U8g2> 2.26.13
|   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |-- <Wire> 1.0.1
|   |-- <U8glib> 1.19.1
|-- <Wire> 1.0.1
|-- <SPI> 1.0
Compiling .pio/build/heltec_wifi_kit_32/src/main.cpp.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_pb32h1.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_pb8h1.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_pb8h1f.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_pb8h2.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_pb8h8.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_pb8v1.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_pb8v2.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_pbxh16.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_pbxh24.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_polygon.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_rect.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_rot.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_scale.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_state.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_u16toa.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_u8toa.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib95b/U8glib_ID7/clib/u8g_virtual_screen.c.o
Compiling .pio/build/heltec_wifi_kit_32/lib3ec/SPI/SPI.cpp.o
Compiling .pio/build/heltec_wifi_kit_32/lib862/Wire/Wire.cpp.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/U8x8lib.cpp.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_bitmap.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_box.c.o
Archiving .pio/build/heltec_wifi_kit_32/lib3ec/libSPI.a
Indexing .pio/build/heltec_wifi_kit_32/lib3ec/libSPI.a
Archiving .pio/build/heltec_wifi_kit_32/lib95b/libU8glib_ID7.a
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_buffer.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_circle.c.o
Indexing .pio/build/heltec_wifi_kit_32/lib95b/libU8glib_ID7.a
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_cleardisplay.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_d_memory.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_d_setup.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_font.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_fonts.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_hvline.c.o
Archiving .pio/build/heltec_wifi_kit_32/lib862/libWire.a
Indexing .pio/build/heltec_wifi_kit_32/lib862/libWire.a
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_input_value.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_intersection.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_kerning.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_line.c.o
Compiling .pio/build/heltec_wifi_kit_32/libd86/U8g2_ID942/clib/u8g2_ll_hvline.c.o
[... 783 lines cut ...]
.pio/libdeps/heltec_wifi_kit_32/ArduinoMenu library_ID1468/src/macros.h:199:3: error: cannot convert 'time_t (*)(time_t*) {aka long int (*)(long int*)}' to 'Menu::prompt*' in initialization
   };\
   ^
.pio/libdeps/heltec_wifi_kit_32/ArduinoMenu library_ID1468/src/macros.h:192:42: note: in expansion of macro 'altMENU'
 #define MENU(id,text,aFn,mask,style,...) altMENU(Menu::menu,id,text,aFn,mask,style,(Menu::_menuData|Menu::_canNav),__VA_ARGS__)
                                          ^
src/main.cpp:139:1: note: in expansion of macro 'MENU'
 MENU(mainMenu, "Main menu", doNothing, noEvent, wrapStyle, OP("Op1", doNothing, noEvent), OP("Op2", doNothing, noEvent)
 ^
src/main.cpp:139:6: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 MENU(mainMenu, "Main menu", doNothing, noEvent, wrapStyle, OP("Op1", doNothing, noEvent), OP("Op2", doNothing, noEvent)
      ^
[... cut ...]
src/main.cpp:139:1: note: in expansion of macro 'MENU'
 MENU(mainMenu, "Main menu", doNothing, noEvent, wrapStyle, OP("Op1", doNothing, noEvent), OP("Op2", doNothing, noEvent)
 ^
*** [.pio/build/heltec_wifi_kit_32/src/main.cpp.o] Error 1
================================================================================== [ERROR] Took 11.76 seconds ==================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

````

neu-rah, 2019-07-19:
@dremeier 

```text
.pio/libdeps/heltec_wifi_kit_32/ArduinoMenu library_ID1468/src/macros.h:210:24: error: 'Menu::menu time' redeclared as different kind of symbol
```

`time´ is a name used by a function, please choose another name

exocode, 2019-07-19:
@neu-rah hm... It's simple the U8g2 example... copy, pasted, then compiled. I didn't renamed anything: https://github.com/neu-rah/ArduinoMenu/blob/master/examples/U8G2/U8G2/U8G2.ino

neu-rah, 2019-07-19:
ahh then its me having a name collision on that board... checking that.

thanks for reporting

exocode, 2019-07-19:
renamed it to "timed"... compiled: worked! Thank you! Took me seven hours to get my first example running... whew... 😬

neu-rah, 2019-07-20:
@dremeier hope this fix it also for due, please reopen if not

closed 2019-07-20 by neu-rah

dremeier, 2019-07-20:
@neu-rah, Yes that's working also for he Due. Thanks

Ulli2k, 2019-08-15:
same issue.
could you please merge the fix?

neu-rah, 2019-08-16:
@Ulli2k I only have warnings because I asked compilation details...