neu-rah/ArduinoMenu #251: Cursor marker

Ulli2k, 2019-07-14:
I got my u8g2 Display working!

Currently the cursor marks the line by changing the background colour.
How can I change it back to have a marker in the front of the line like:
```
   Op1
> Op2
   OP3
```

Additionally I do not understand who to configure the following struct. Is there a HowTo?
```
  const colorDef<uint8_t> colors[] MEMMODE={ //  {{disabled normal,disabled selected},{enabled normal,enabled selected, enabled editing}}
    {{0,0},{0,1,0}},//bgColor
    {{0,0},{1,1,1}},//fgColor
    {{1,1},{1,0,0}},//valColor
    {{1,1},{1,0,0}},//unitColor
    {{0,0},{0,0,1}},//cursorColor
    {{1,1},{1,0,0}},//titleColor
  };
```

neu-rah, 2019-07-14:
>Currently the cursor marks the line by changing the background colour.
How can I change it back to have a marker in the front of the line like

only by changing the u8g2 driver to print as text style and yet use u8g2 print functions

>Additionally I do not understand who to configure the following struct. Is there a HowTo?

They should be on wiki... however comments might be helpful,

bgColor - definition of background colors (see line description bellow)  
fgColor - foreground colors  
valColor - field value  
unitColor - field value unit  
cursorColor - cursor  
titleColor - menu title  

each line contains values in two groups for disabled and enabled items

first group, disabled item colors   
{disabled normal,disabled selected}

second group, enabled item colors for use when normal, selected or editing  
{enabled normal,enabled selected, enabled editing}

the above table show a mono-chrome color table (colors only 0 or 1) but the color format depends on the device, always on the same order.

note: some combinations might have no expression on some drivers

Ulli2k, 2019-07-18:
Ok, therefore the u8g2 driver is only supporting the cursor by changing the backgroundcolor of the row.

I already read the comments. But I do not understand way I can disable the item color bis to values.
(1,0) (.....) disable in normal mode
(....) (0....) disable in normal mode
Am I right?

neu-rah, 2019-07-18:
no, you do not disable the item color, that color definition is for the menu system to use when drawing disable**d** item, items are enabled/disabled by the code