neu-rah/ArduinoMenu #253: KeyIn with interrupts

wayne2000, 2019-07-27:
Thanks for nice menu lib! I am able to use it now in a non-Ardunio env on STM32.
Now I am thinking to make it work within FreeRTOS. One thing confusing me is how to make use of GPIO interrupts to make the menu codes run more effeciently (less CPU %). I saw in Class KeyIn the peek() calls digitalRead, but with it we have to do regular poll on GPIO, instead of relying on interrupts. Anything I missed or suggestion? Thanks in advance!

neu-rah, 2019-07-27:
Hi!

this encoder driver uses interrupts so it might be inspiring https://github.com/neu-rah/ArduinoMenu/blob/5936fb41caf3c7f776d5d1be6138a9b0f16664f2/src/menuIO/encoderIn.h

it uses this library  https://github.com/neu-rah/PCINT  
but can be made use others

wayne2000, 2019-07-31:
Thanks Neu-rah. Maybe I didn't get the question clear. 
I want to have one task in FreeRTOS for navRoot::doInput(), but don't want to have periodic GPIO poll inside, instead want to have this task suspend for most of time until some key is pressed, so that the task can be woken up by interrrupts.

Finally I made it by sending taskNotify from GPIO interrupt callback, then in one task calling navRoot::doNav() with navigating commands and doOuput().

closed 2019-07-31 by wayne2000