neu-rah/ArduinoMenu #254: example for esp32 with rotary encoder?

yentzee, 2019-08-25:
Hi,

is there any example using an esp32 with a rotary encoder and a tft?
I looked for ages but couldnt find anything that would work on my setup.
cableing is correct. I can use the display but my skills are almost zero so i am not able to edit to my needs.
If I had a working example of a menu I could modify it to my needs.

Everything pretty complicated :(

neu-rah, 2019-08-27:
Hi, you are right this is a missing spot... anyone?

Joennuh, 2019-08-28:
I've rotating encoders but so far never experimented with it. Currently Ialso don't have the time to experiment with. Hope to see an example soon or else I will experiment with it somewhere in the future.

nathandarnell, 2019-09-03:
I have a working sketch with a rotary encoder and TFT and just submitted a pull request.  I think I did this correctly but maybe not?  Hardest parts have been correctly wiring and defining the pins in TFT_eSPI, having a rotary encoder that works well, setting up the ESP32 timer, and turning off doubleclicks in the ClickEncoder library so keyIn isn't needed.

neu-rah, 2019-09-05:
Thanks all. Just for reference, solved by: https://github.com/neu-rah/ArduinoMenu/pull/256

closed 2019-09-05 by neu-rah