neu-rah/ArduinoMenu #258: Compile error at U8G2 sample code related with colors const definition

jjgarciajz, 2019-10-01:
Hello,
I'm testing your library using a Pro Micro & 128x32 SSD1306 I2C display.

I took the U8G2 sample code provided with the library. Changed the u8g2 constructor to the proper display, set rotary encoder pins properly, etc... But I'm getting a compile error related with the const colors definition.

```C:\xxxxx\Documents\Arduino\libraries\ArduinoMenu_library\src/menuIO/u8g2Out.h: In function 'global constructors keyed to 65535_0_MenuLibraryDemo.ino.cpp.o':
MenuLibraryDemo:83:25: error: variable 'colors' with dynamic initialization put into program memory area
 const colorDef<uint8_t> colors[] MEMMODE={
                         ^
lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status```

Am I doing something wrong?
Thanks.

neu-rah, 2019-10-01:
Hi!
is that, color definition inside a function? because it should be at global scope.

jjgarciajz, 2019-10-01:
Hello! No, it's in the global scope.

Check your sample file, line 81
https://github.com/neu-rah/ArduinoMenu/blob/master/examples/U8G2/U8G2/U8G2.ino

neu-rah, 2019-10-01:
yes, confirmed
its ok with platformio but with that error on arduino-ide, please add the number 6 to array dimension
like this:
```c++
const colorDef<uint8_t> colors[6] MEMMODE={
```

i will submit the correction on next push, thanks for reporting it.

jjgarciajz, 2019-10-01:
Ah! I should've noticed that. Now it is compiling correctly.
You're welcome! Thanks for solving it :)

closed 2019-10-01 by jjgarciajz