neu-rah/ArduinoMenu #259: ESP32 & Rotary encoder for LCD I2C?

graziano71, 2019-10-05:
Hello!
Kind help needed to reuse and adapt example code:
 
https://github.com/neu-rah/ArduinoMenu/pull/256

to change output from TFT into LiquidCrystal Library I2C.

Thanks in advance for any help you may provide!

neu-rah, 2019-10-06:
you can start with PCF8574Out example https://github.com/neu-rah/ArduinoMenu/blob/master/examples/LCDs/PCF8574/PCF8574/PCF8574.ino

and then port the input definition from esp32 encoder example https://github.com/neu-rah/ArduinoMenu/blob/master/examples/ESP32/ClickEncoderTFT.ino

relevant parts

```c++
// Declare the clickencoder
// Disable doubleclicks in setup makes the response faster.  See: https://github.com/soligen2010/encoder/issues/6
ClickEncoder clickEncoder = ClickEncoder(encA, encB, encBtn, encSteps);
ClickEncoderStream encStream(clickEncoder, 1);

// ESP32 timer thanks to: http://www.iotsharing.com/2017/06/how-to-use-interrupt-timer-in-arduino-esp32.html
// and: https://techtutorialsx.com/2017/10/07/esp32-arduino-timer-interrupts/
hw_timer_t *timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;
```
and then replace input definition and link it to navroot

```c++
serialIn serial(Serial);
MENU_INPUTS(in, &encStream, &serial); // &encButton,
NAVROOT(nav, mainMenu, MAX_DEPTH, in, out);
```
dont forget setup initialization
```c++
  clickEncoder.setAccelerationEnabled(true);
  clickEncoder.setDoubleClickEnabled(false); // Disable doubleclicks makes the response faster.  See: https://github.com/soligen2010/encoder/issues/6

  // // ESP32 timer
  timer = timerBegin(0, 80, true);
  timerAttachInterrupt(timer, &onTimer, true);
  timerAlarmWrite(timer, 1000, true);
  timerAlarmEnable(timer);
```
hope i'm not missing parts, i've not compiled this

neu-rah, 2019-10-10:
no special reason other than I had working prototypes using ClickEncoder and therefor was easier to make examples out of them.
And yes the internal encoder driver depends on PCINT (that should be at least noted on the source code, sorry if not). I've never tried PCINT on esp, it should work in theory because other than AVR's, that use a PCINT table all other MCUs AFAIK support interrupts on all pins so the PCINT is just deferring the handling and provide a consistent interface as with AVR, but one can achieve the same without it,

closed 2019-10-17 by neu-rah

chamroeunbm, 2021-03-01:
@neu-rah both examples that mentioned not working on my esp32. i mean the rotary switch only detect the click botton-rotation not detected. run this code on arduino mega and it works fine.
while using this library, my hardwarw works https://github.com/igorantolic/ai-esp32-rotary-encoder

neu-rah, 2021-03-04:
**that's the point**, use that library and issue menu commands to move up/down as intended...
and remove all menu encoder definitions as they will only cause a mess