neu-rah/ArduinoMenu #260: encoder pcint.h is missing

valsesro, 2019-10-22:
When trying to compile lcd example, I get this error C:\SPB_DATA\.platformio\lib\ArduinoMenu library_ID1468\src/menuIO/encoderIn.h:13:56: fatal error: pcint.h: No such file or directory . I tried to install AnotherPcintLibrary that is meant in encoderIn.h but get same arror. Files in AnotherPcintLibrary ara named YetAnotherPcInt so It doesn't correspond to pcint.h used in menu library.

valsesro, 2019-10-22:
When I changed the names of AnotherPcintLibrary to pcint.h, there is another problem 'PCattachInterrupt' was not declared in this scope and 'mixHandler' was not declared in this scope

neu-rah, 2019-10-22:
what board and IDE are you using?
if on arduino IDE install the libs using the "library manager"

neu-rah, 2019-10-30:
we are talking of this https://github.com/neu-rah/PCINT library, right?

valsesro, 2019-11-06:
Thanks, I accidentally used another library, but when I installed provided librarry I always get error lib\PCINT-master\src\pcint.h:62:56: error: 'pcintPinMap' was not declared in this scope

valsesro, 2019-11-06:
Ok, I got it to work, defining pin map for AtMega32U4. Thank you for all your replies.

closed 2019-11-06 by valsesro