neu-rah/ArduinoMenu #263: Can't properly invert colors on a adafruitgfx display

Sh4d, 2019-11-19:
The example code has the following which renders black text on a white background:

```
const colorDef<uint16_t> oled_inverted[6] MEMMODE={
  {{WHITE,BLACK},{WHITE,BLACK,BLACK}},//bgColor
  {{BLACK,WHITE},{BLACK,WHITE,WHITE}},//fgColor
  {{BLACK,WHITE},{BLACK,WHITE,WHITE}},//valColor
  {{BLACK,WHITE},{BLACK,WHITE,WHITE}},//unitColor
  {{BLACK,WHITE},{WHITE,WHITE,WHITE}},//cursorColor
  {{BLACK,WHITE},{WHITE,BLACK,BLACK}},//titleColor
};
```

I flipped this around to white text on black background:

```
const colorDef<uint16_t> oledColors[6] MEMMODE={
  {{BLACK,WHITE},{BLACK,WHITE,BLACK}},//bgColor
  {{WHITE,BLACK},{WHITE,BLACK,WHITE}},//fgColor
  {{WHITE,BLACK},{WHITE,BLACK,WHITE}},//valColor
  {{WHITE,BLACK},{WHITE,BLACK,WHITE}},//unitColor
  {{WHITE,BLACK},{WHITE,BLACK,WHITE}},//cursorColor
  {{WHITE,BLACK},{BLACK,BLACK,WHITE}},//titleColor
};
```

However the border around the selected / unselected items doesn't seem to be configurable, and so I'm ending up with a white border around all the items (rather than black).

neu-rah, 2019-11-19:
can you try this one?

```c++
const colorDef<uint16_t> oled_inverted[6] MEMMODE={
  {{BLACK,WHITE},{BLACK,WHITE,WHITE}},//bgColor
  {{WHITE,BLACK},{WHITE,BLACK,BLACK}},//fgColor
  {{WHITE,BLACK},{WHITE,BLACK,BLACK}},//valColor
  {{WHITE,BLACK},{WHITE,BLACK,BLACK}},//unitColor
  {{WHITE,BLACK},{BLACK,BLACK,BLACK}},//cursorColor
  {{WHITE,BLACK},{BLACK,WHITE,WHITE}},//titleColor
};
```

Sh4d, 2019-11-19:
Damn, I thought I'd tried everything but that worked. Thanks!

closed 2019-11-19 by neu-rah