neu-rah/ArduinoMenu #264: Buttons for changing numeric value in a field are reversed

Sh4d, 2019-11-19:
Using:

```
// Map physical buttons to joystick
keyMap joystickBtn_map[]={
 {-BTN_SEL, defaultNavCodes[enterCmd].ch} ,
 {-BTN_UP, defaultNavCodes[upCmd].ch} ,
 {-BTN_DOWN, defaultNavCodes[downCmd].ch}  ,
};
keyIn<3> joystickBtns(joystickBtn_map);
```

This is working as expected to navigate the menus, however when I select a field value the UP button decrements it and the DOWN button increments it:

```
MENU(mainMenu, "Main menu", doNothing, noEvent, wrapStyle
  , SUBMENU(subModes)
  , SUBMENU(subPalette)
  , FIELD(curSpeed , "Speed", "", 1, 1000, 100, 10, doNothing, noEvent, wrapStyle)
  , SUBMENU(subSettings)
  , SUBMENU(subAbout)
  , EXIT("Back")
  );

NAVROOT(nav, mainMenu, MAX_DEPTH, joystickBtns, out);
```

neu-rah, 2019-11-19:
yes, it defaults to encoder, please add:

```c++
options->invertFieldKeys=true;
```

to your setup function

Sh4d, 2019-11-19:
Argh, I saw that option but I coulda sworn it said it was deprecated so I didn't try it. Success, thanks again for the quick response!

closed 2019-11-19 by neu-rah