neu-rah/ArduinoMenu #266: altMENU shows 0 value but expected saved value

vikulin, 2019-12-04:
altMENU shows 0 value but expected saved value. The code is following:

```
//define a pad style menu (single line menu)
//here with a set of fields to enter a time in hh:mm format
altMENU(menu,timeMenu,"Time",doNothing,noEvent,noStyle,(systemStyles)(_asPad|Menu::_menuData|Menu::_canNav|_parentDraw)
  ,FIELD(hrs,"",":",0,23,1,0,doNothing,noEvent,noStyle)
  ,FIELD(mins,"","",0,59,10,1,doNothing,noEvent,wrapStyle)
);
```

In output I see **Time   0:   0** but expected non zero values which stored in hrs and mins variables

Could you please advise how to archive this?

neu-rah, 2019-12-04:
can you try something like this:

```c++
PADMENU(birthDate,"Birth",doNothing,noEvent,noStyle
  ,FIELD(year,"","/",1900,3000,20,1,doNothing,noEvent,noStyle)
  ,FIELD(month,"","/",1,12,1,0,doNothing,noEvent,wrapStyle)
  ,FIELD(day,"","",1,31,1,0,doNothing,noEvent,wrapStyle)
);
```

from example: https://github.com/neu-rah/ArduinoMenu/blob/537427d5165493f8b0e0ea24ec515aa4e90ae35c/examples/Serial/serialio/serialio/serialio.ino#L98

vikulin, 2019-12-05:
I have tryed PADMENU and it work as expected. But I need to save the year,
month, day variables in EEPROM after change. How can I do this?

vikulin, 2019-12-05:
Thank you for so quick response.

чт, 5 дек. 2019 г. в 20:57, Vadym Vikulin <vadym.vikulin@gmail.com>:

neu-rah, 2019-12-05:
you can provide and event handler function to be called whenever and event corresponds to an also provided event mask. On above code we have a system provided event handler `doNothing` and a mask `noEvent` that prevents it to be called, just change that.

accepted function formats are explained here: https://github.com/neu-rah/ArduinoMenu/wiki/Actions

event enumeration is here: https://github.com/neu-rah/ArduinoMenu/blob/d2ca6505f76e8b4cf4ce3b3667f2dd9968f1d3c3/src/menuBase.h#L115

note that (by compatibility reasons) by default the enter event will be sent whenever the value changes, if that is not desirable, as kit should not in case of eeprom write, you can enable the update event on the navigation root object, after that and update event is sent when the value changes and an enter event is obly send when the edit terminates, please also note that it will be sent even if the edit terminates with and escape... just because that the value remains as edited independent of termination done by enter or esc. 

This is done so because the target value could be a motor PWM and sudden changes can have bad results... there is however a plugin with a field that restores its value when edit is terminated by escape (https://github.com/neu-rah/ArduinoMenu/blob/master/src/plugin/cancelField.h)

neu-rah, 2019-12-06:
please open another issue if not solved as we are already off-topic

closed 2019-12-06 by neu-rah

vikulin, 2019-12-06:
Thank you. It works for me. I have added working script to Github:

https://github.com/vikulin/LCD_LAT